canopen-python/canopen #1: Cannot parse $NODEID

tuwuhs, 2016-10-27:
Tried to create a Copley drive node from [EDS file](http://www.copleycontrols.com/Motion/zip/CANopen-EDS.zip), got error when parsing lines with ```$NODEID``` literal in it.
Problematic lines (CopleyAmp.eds):
```
[1200sub1]
ParameterName=SDO COB-ID Client->Server
ObjectType=7
DataType=7
AccessType=RO
PDOMapping=0
DefaultValue=$NODEID+1536
LowLimit=$NODEID+1536
HighLimit=$NODEID+1536

```

Error:
```
In [4]: node = network.add_node(6, './CopleyAmp.eds')
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-4-b30d33f7eaa4> in <module>()
----> 1 node = network.add_node(6, './CopleyAmp.eds')

/home/tuwuhs/.local/lib/python2.7/site-packages/canopen/network.pyc in add_node(self, node, object_dictionary)
     87         """
     88         if isinstance(node, int):
---> 89             node = Node(node, object_dictionary)
     90         node.network = self
     91         self.nodes.append(node)

/home/tuwuhs/.local/lib/python2.7/site-packages/canopen/node.pyc in __init__(self, node_id, object_dictionary)
     32 
     33         if object_dictionary:
---> 34             self.set_object_dictionary(object_dictionary)
     35 
     36         self.sdo = SdoClient(self, node_id)

/home/tuwuhs/.local/lib/python2.7/site-packages/canopen/node.pyc in set_object_dictionary(self, object_dictionary)
     63         if not isinstance(object_dictionary,
     64                           objectdictionary.ObjectDictionary):
---> 65             object_dictionary = objectdictionary.import_od(object_dictionary)
     66         self.object_dictionary = object_dictionary
     67 

/home/tuwuhs/.local/lib/python2.7/site-packages/canopen/objectdictionary/__init__.pyc in import_od(filename)
     38     if filename.endswith(".eds"):
     39         from . import eds
---> 40         return eds.import_eds(filename)
     41     elif filename.endswith(".epf"):
     42         from . import epf

/home/tuwuhs/.local/lib/python2.7/site-packages/canopen/objectdictionary/eds.pyc in import_eds(filename)
     52             entry = od[index]
     53             if isinstance(entry, objectdictionary.Record):
---> 54                 var = build_variable(eds, section, index, subindex)
     55                 entry.add_member(var)
     56 

/home/tuwuhs/.local/lib/python2.7/site-packages/canopen/objectdictionary/eds.pyc in build_variable(eds, section, index, subindex)
     64     var.access_type = eds.get(section, "AccessType")
     65     if eds.has_option(section, "LowLimit"):
---> 66         var.min = int(eds.get(section, "LowLimit"), 0)
     67     if eds.has_option(section, "HighLimit"):
     68         var.max = int(eds.get(section, "HighLimit"), 0)

ValueError: invalid literal for int() with base 0: '$NODEID+1536'
```

christiansandberg, 2016-10-27:
Oh, that's right. Would it be OK to just ignore everything not directly parsable as integers? Currently I've tried not to assume the node id while initializing the node, but maybe it could be assumed to never change.

christiansandberg, 2016-10-27:
You're welcome to suggest a more long-term solution on how to handle $NODEID if you need it.

closed 2016-10-27 by christiansandberg

tuwuhs, 2016-10-27:
Yes the short-term solution would work, thanks. It's not really essential, but I think it is quite common to have $NODEID in .eds files.

> Currently I've tried not to assume the node id while initializing the node, but maybe it could be assumed to never change.
In my opinion we can assume a node id at initialization (and so it can be passed to the object dictionary parser), and when we need to have another node id we can always create a new node. What is the specific situation where the node id is changed after being initialized?

christiansandberg, 2016-10-28:
I have been thinking about doing this change to simplify unit testing anyway, but haven't got around to do it yet. Then it would be possible to add evaluation of $NODEID at parse time.

tuwuhs, 2016-10-28:
Let me know if I can help. :wink:

commit 016fdaf7 (2016-10-27): Do not fail when parsing min and max values.

Fixes #1.