canopen-python/canopen #8: Receiving

Muschasgrazias, 2017-01-17:
Hello Christian,

I'am working in a CANOpen project myself and found your repository. Great work btw!

Its not really an Issue with your code or anything I'am just new to Github and not sure where to put my request.

I would like to implement a thread based receiving (classic client - server based). Therefore no messages will be lost and there is no need to wait for a reception. 

Did you consider something like this? Then I could contribute via a branch if you like.

Best regards

christiansandberg, 2017-01-17:
Hi! I'm not sure what you mean? Are you referring to the reception of PDOs? In that case I recently added a [`add_callback()`](http://canopen.readthedocs.io/en/stable/pdo.html#canopen.pdo.Map.add_callback) method per PDO which you can use to get notified on new incoming messages.

Muschasgrazias, 2017-01-18:
Hi,
Kind of...
My idea is to have one blocking thread, his only job is to receive any messages (PDOs, SDOs, Syncs, EMCYs and so on). 
Spawns worker threads for each message that is received 
The worker thread now decides from the COB-ID what kind of message it is and saves the message in a dictionary for the corresponding type (the key is the ID)
So for every message type you have a dictionary that can then be accessed via the main program.

Pros: You can be 100% sure you get every message on the bus
Any waiting in the main program doesn't disrupt receiving (i.e.: wait for acknowledge SDO)
If you have 127 nodes (max for 11 bit ID) you don't need to add for every node a callback

Cons: Maybe overkill?!
A lot of dictionaries
Rewriting some code necessary (also not sure)

Best regards

christiansandberg, 2017-01-18:
I think it already is working approximately like that if I understand your idea correctly.

When nodes are added to the network, they subscribe to specific CAN-IDs, like 0x586 for SDO responses to node 6 and so on.
On `connect()` we start one background thread which takes care of waiting for new messages on the bus and when one is received, it checks if there are any subscriptions for that CAN-ID and feeds it to that service for processing. Then it goes back to checking for new messages.
This is done through the python-can Notifier and Listener classes so it is not directly obvious that there is a thread.

christiansandberg, 2017-01-18:
One thing I've been thinking of is if there should be an asynchronous API as well so that it is possible to start an SDO operation but reading the result at a later occasion. This can however be accomplished today at a higher level, using e.g. the [concurrent.futures](https://docs.python.org/3/library/concurrent.futures.html) package.

Muschasgrazias, 2017-01-18:
Ah, I'am sorry I didn't know that this is already implemented one layer lower. I didn't have the time to go through the pcan-can library.

So adding a callback for EMCY messages would be the same as for the PDO one?

One last thing: Object Dictionary
I added a value that is an unsigned integer 8 that ranges from 0 - 6
In the EDS file stands now this:
`[OptionalObjects]
SupportedObjects=1
1=0x00A0

[A0]
ParameterName=UNSIGNED8[0-6]
ObjectType=0x9
SubNumber=4

[A0sub0]
ParameterName=Number of Entries
ObjectType=0x7
DataType=0x0005
AccessType=ro
DefaultValue=3
PDOMapping=0

[A0sub1]
ParameterName=Type
ObjectType=0x7
DataType=0x0005
AccessType=ro
DefaultValue=5
PDOMapping=0

[A0sub2]
ParameterName=Minimum Value
ObjectType=0x7
DataType=0x0005
AccessType=ro
DefaultValue=0
PDOMapping=0

[A0sub3]
ParameterName=Maximum Value
ObjectType=0x7
DataType=0x0005
AccessType=ro
DefaultValue=6
PDOMapping=0

[2000]
ParameterName=value
ObjectType=0x7
DataType=0x00A0
AccessType=rw
DefaultValue=0
PDOMapping=1
`
And if I want to set this value that is on Index 0x2000 sub 0x0 then I get this error:
`TypeError: Do not know how to encode 1 to data type A0h`

My question now is: Shouldn't this be possible, because I found the encode_raw function where the range check with the datatype should be done. I guess that the datatype pointer is not converted into the optional object. Do you know what I should change?

Thanks for answering this fast!

Have a nice day

christiansandberg, 2017-01-18:
I haven't added callbacks to EMCYs yet but that I will definitely implement that as well.

Datatype 0xA is OCTET_STRING, maybe you meant to use 0x5 which is UNSIGNED8?

Muschasgrazias, 2017-01-18:
In eds.py I added a function in build_variable to select the optional objects parameters.

I'am not sure if its worth patching it in, because I'am using the object dictionary editor from CANFestival. I guess you use the eds editor from vector? 

Or do you have something else because you wrote in the read me example program ['Producer heartbeat time'] even though all my editors want it in upper case like ['Producer Heartbeat Time']

christiansandberg, 2017-01-19:
I would be happy to look at your solution if it is general enough.

I've used Vector's tool to generate an EDS but I don't think there are any rules on how to name the items in the object dictionary.

Muschasgrazias, 2017-01-20:
Hi,

Probably not worth for a pull request, but this is my fast, dirty fix for now.
Missing are the min - max values that are also written in the eds file
```
def build_variable(eds, section, index, subindex=0):
    name = eds.get(section, "ParameterName")
    var = objectdictionary.Variable(name, index, subindex)
    var.data_type = int(eds.get(section, "DataType"), 0)
    var.access_type = eds.get(section, "AccessType").lower()
    if var.data_type > 0x1B:
        # The object dictionary editor from CANFestival creates an optional object if min max values are used
        # This optional object is then placed in the eds under the section [A0] (start point, iterates for more)
        # The eds.get function gives us 0x00A0 now convert to String without hex representation and upper case
        # The sub2 part is then the section where the type parameter stands
        var.data_type = str(hex(int(eds.get(section, "DataType"), 0))).replace("0x", "").upper() + "sub2"
        var.data_type = int(eds.get(var.data_type, "DataType"), 0)

    if eds.has_option(section, "LowLimit"):
        try:
            var.min = int(eds.get(section, "LowLimit"), 0)
        except ValueError:
            pass
    if eds.has_option(section, "HighLimit"):
        try:
            var.max = int(eds.get(section, "HighLimit"), 0)
        except ValueError:
            pass
    return var
```

Kind regards

christiansandberg, 2017-01-20:
I did some simplification of your code which I hope will still work. Also shouldn't it be A0sub1 defining the type?

Muschasgrazias, 2017-01-25:
Hi,
Yeah you are right. We should take A0sub1 and the default value.
I just liked that the eds.get function has as parameter DataType (more understandable?)
But of course if someone doesn't use minimum and maximum values this won't work properly.

Best regards

closed 2017-10-30 by Muschasgrazias