canopen-python/canopen #9: Bit mapping in PDOs

Muschasgrazias, 2017-01-19:
Hi Christian,

Sorry I'am so annoying, but I have again an Issue.

Bit mapping for PDOs isn't implemented right?

Best regards

christiansandberg, 2017-01-19:
Yes that's right, currently it only supports data types that are whole bytes. I wasn't sure how widely used other types were. Maybe you could try to implement it? The bitstruct package could perhaps help.

Muschasgrazias, 2017-01-20:
Hi,

Yes I'am on it right now.
The thing is I'am coming from C and for me its easier to just shift and mask the bits.
But thank you for the tip with the bitstruct, I'am going to look into it :)

Another thing I would like to ask is, if you thought about implementing a function that reads out the dictionary of the slave node and uses that as dictionary object. I'am not sure now if there is a mechanic in the CiA standard but maybe you know more!

Best regards

Muschasgrazias, 2017-01-20:
First Prototyp:

```
class Variable(common.Variable):
    """One object dictionary variable mapped to a PDO."""

    def __init__(self, od):
        self.msg = None
        #: Location of variable in the message in bits
        self.offset = None
        self.name = od.name
        self.length = None
        if isinstance(od.parent, (objectdictionary.Record,
                                  objectdictionary.Array)):
            self.name = od.parent.name + "." + self.name
        common.Variable.__init__(self, od)

    def get_data(self):
        byte_offset = self.offset // 8
        bit_offset = self.offset % 8
        data = int.from_bytes(self.msg.data[byte_offset:byte_offset + len(self.od) // 8], byteorder='little', signed=False)
        data = (data >> bit_offset) & (2**self.length-1)

        return data.to_bytes(len(self.od)//8, byteorder='little')

    def set_data(self, data):
        byte_offset = self.offset // 8
        logger.debug("Updating %s to %s in message 0x%X",
                     self.name, binascii.hexlify(data), self.msg.cob_id)
        self.msg.data[byte_offset:byte_offset + len(data)] = data](url)
```

This is the "new" variable class with a length parameter that gets set here:

```
    def read(self):
        """Read PDO configuration for this map using SDO."""
        cob_id = self.com_record[1].raw
        self.cob_id = cob_id & 0x7FF
        logger.info("COB-ID is 0x%X", self.cob_id)
        self.enabled = cob_id & PDO_NOT_VALID == 0
        logger.info("PDO is %s", "enabled" if self.enabled else "disabled")
        self.rtr_allowed = cob_id & RTR_NOT_ALLOWED == 0
        logger.info("RTR is %s", "allowed" if self.rtr_allowed else "not allowed")
        self.trans_type = self.com_record[2].raw
        logger.info("Transmission type is %d", self.trans_type)

        self.map = []
        offset = 0
        for entry in self.map_array.values():
            if entry.od.subindex == 0:
                continue
            value = entry.raw
            index = value >> 16
            subindex = (value >> 8) & 0xFF
            size = value & 0xFF
            if size == 0:
                continue
            var = self._get_variable(index, subindex)

            #assert size == len(var.od), "Size mismatch"
            var.offset = offset
            var.length = size
            logger.info("Found %s (0x%X:%d) in PDO map",
                        var.name, index, subindex)
            self.map.append(var)
            offset += size
        self._update_data_size()

        if self.enabled:
            self.pdo_node.network.subscribe(self.cob_id, self.on_message)
```

Best regards

christiansandberg, 2017-01-20:
This is a good start! But `int.from_bytes` and `int.to_bytes` do not exist in Python 2.7 which should preferably be supported as well.

Muschasgrazias, 2017-01-25:
Hi,

I read a few stackoverflow comments about the whole bytearray into int thing.
As far is I understood:
Python 2.x has decode and encode for byte arrays
Pythont 3.x has from_bytes and to bytes (doesn't support decode and encode anymore)
To be compatible for both is quite troublesome and you have to do a lot of conversions especially because we have to shift and mask the bytes and then transform into little endian again.
I'am sorry, but I'am only programming on python 3,x and I don't really know what changed.

Best regards

christiansandberg, 2017-01-25:
I'm fine with a temporary solution where variables which are not multiples of 8 bits only work in Python 3 for now, but "normal" variables must still work in Python 2 like today.

It's a tricky thing to solve, so that's why I think using a library like bitstruct would offload the heavy lifting from this library.

Muschasgrazias, 2017-01-26:
I have more time this week.
I will try to solve it :)

Muschasgrazias, 2017-05-17:
Hi,

Had some time to spare and wrote the function for python 2 and 3:
```
import struct
    def get_data(self):
        byte_offset = self.offset // 8
        bit_offset = self.offset % 8
        data = self.msg.data[byte_offset:byte_offset + len(self.od) // 8]
        length = len(data)
        if length == 1:
            length = 'B'
        elif length == 2:
            length = 'H'
        elif length == 4:
            length = 'L'
        elif length == 8:
            length = 'Q'
        data = struct.unpack("<"+length, data)[0]
        data = (data >> bit_offset) & (2**self.length-1)
        data = struct.pack("<"+length, data)
        return data
```
This worked for me.

Best regards

christiansandberg, 2017-05-18:
It won't work for signed data types, but it's a start. I'll try to implement it.

Muschasgrazias, 2017-07-27:
Hello Christian,

Wouldn't the future import suffice?

```
from __future__ import unicode_literals
    def get_data(self):
        byte_offset = self.offset // 8
        bit_offset = self.offset % 8
        data = int.from_bytes(self.msg.data[byte_offset:byte_offset + len(self.od) // 8], byteorder='little', signed=False)
        data = (data >> bit_offset) & (2**self.length-1)

        return data.to_bytes(len(self.od)//8, byteorder='little')
```
Then we don't need the struct part and its still python 2 and 3 conform.

Best regards

Muschasgrazias, 2017-10-24:
Hey,

Signed isn't used in this context right?
The actual decoding for the data types is done from the OD entries which is in the hierarchy above this level.  `def decode_raw(self, data):`
This function just extracts the correct amount of bits (without modifying it) and returns them to an upper level. 

Best regards

christiansandberg, 2017-10-24:
The problem is that the higher unused bits will be filled with zeroes which is fine for unsigned values, but for signed values these should be filled with ones so that the data gets converted correctly.

Say you have a 4-bit signed data type. If the value is -1 the data is encoded as 0xF which will be converted to 0x0F by get_data() and then to 15.

Muschasgrazias, 2017-10-24:
Hey,

Yeah sorry my fault. How is this:

```
        data = self.msg.data[byte_offset:byte_offset + len(self.od) // 8]
        od_type = self.od.STRUCT_TYPES[self.od.data_type].format
        import struct
        data = struct.unpack(od_type, data)[0]
        data = (data >> bit_offset) & (2**self.length-1)
        data = struct.pack(od_type, data)
        return data
```

Best regards

Muschasgrazias, 2017-10-30:
#48

Muschasgrazias, 2017-10-30:
#48

closed 2018-02-23 by Muschasgrazias