canopen-python/canopen #10: Bug in node without EDS file

philschenker, 2017-02-01:
When I try to run the following code:

```
import canopen
network = canopen.Network()
node = network.add_node(45, object_dictionary=None)
network.connect(channel=0, bustype="ixxat", can_filters=None)
node.sdo[0x1018][2].raw
network.disconnect()
```

I get an error:

`No handlers could be found for logger "canopen.objectdictionary.eds"`

If I attach your sample.eds, everything works.

I think there's something wrong in eds.py on line 42

- I'm working with version 0.4.0 from pip
- tried with 0.5.0.dev1

christiansandberg, 2017-02-01:
Right now there is no "default" object dictionary, not even for the mandatory parts, so something has to be provided in order to use SDOs. I have tried to implement reading the EDS from the node if one isn't specified, but I can't test it.

Do you get any other messages than the one you've written? I don't think it's an actual error, just information.

philschenker, 2017-02-01:
Hello Christian, thanks for your answer.

I think the SDO 0x1021 isn't implemented in the canopen devices I'm testing. But what I was hoping is, if it fails to read out the EDS file it still should get the ability to access SDO's with e.g.

`node.sdo[0x1000].raw`

But that is unfortunately not working. It works though when I'm giving any EDS file that isn't matching my device and then accessing with the code above.

christiansandberg, 2017-02-02:
The thing is that the library must know how to interpret the data, such as if it is signed or unsigned. You can actually read raw data without an object dictionary, but you have to convert the bytes to something meaningful yourself:
data = node.sdo.upload(0x1000, 0)

philschenker, 2017-02-09:
Hey Christian, thank you for the clarification. You're right this is not a bug :-).
Thank you!

closed 2017-02-09 by philschenker