canopen-python/canopen #11: Request: Configurable SDO timeout or network enumeration feature

igutekunst, 2017-02-02:
We use the following code to enumerate CANOpen devices on the network

```python        
network = range(1,128)
enumerated_devices = []
for id, p in enumerate(network):
    try:
        node = network[id]
        node.nmt.state = 'PRE-OPERATIONAL'
        dtype = node.sdo.upload(0x1000, 0)
        if id not in self.enumerated_pucks:
            enumerated_devices.append(id)
    except Exception as e:
        pass

# Do stuff with enumerated_devices
```

The current timeout makes this process rather slow. How would you advice we enumerate the network faster?

In an earlier version, I changed sdo.py to only retry 1, and set the timeout to 0.1 seconds, which was good enough for testing. However, we'd like to do it the correct way.

christiansandberg, 2017-02-03:
I implemented a [node scanner](http://canopen.readthedocs.io/en/latest/network.html#canopen.network.NodeScanner) which perhaps could fill your needs:
```python
network.scanner.search()
# Might need to sleep a short time here
network.scanner.nodes
```

Otherwise you can set the [canopen.sdo.SdoClient.MAX_RETRIES](http://canopen.readthedocs.io/en/latest/sdo.html#canopen.sdo.SdoClient.MAX_RETRIES) and [canopen.sdo.SdoClient.RESPONSE_TIMEOUT](http://canopen.readthedocs.io/en/latest/sdo.html#canopen.sdo.SdoClient.RESPONSE_TIMEOUT), either for the whole class or per node.

igutekunst, 2017-02-03:
Perfect. I didn't look through the documentation well enough.

closed 2017-02-03 by igutekunst

christiansandberg, 2017-02-03:
I haven't documented it well enough either. :)

igutekunst, 2017-02-06:
I busted used the node scanner, and it is so much faster! This totally made my day 👍