canopen-python/canopen #13: pdo read returns nil, no map loaded

leongzhang, 2017-02-07:
Hi, first thank you for the package, it saved a lot time dealing with the dictionary manually. 

i'm using a device from datexel.it (http://www.datexel.it/products/CANOPEN/DAT7016) to get thermocouple data. according to the datasheet, there is one tpdo from the unit. but after calling the pdo.read() method the pdo.tx or pdo.rx were all empty. I wonder if the tx and rx are supposedly to read from the eds file (object dictionary) or from an sdo command return? i wonder if this is a eds issue or i'm missing something here.

i've got the sdo part to work after implementing the connect/disconnect/send_message with a third party CAN driver. 

sorry im new to githut too.. let me know if this is the proper place to post such questions. 

regards
gz

christiansandberg, 2017-02-07:
Does the EDS file have index 0x1A00 defined? It will be used to read the PDO configuration using SDO.

You can get more debug output to see what it's doing by increasing the log level to debug.

```python
import logging
logging.basicConfig(level=logging.DEBUG)
import canopen
...
```

christiansandberg, 2017-02-07:
I looked at the EDS file and for some reason it only has the 2nd PDO defined. My implementation starts with looking for the 1st PDO, then 2nd and so on. When it does not find the PDO it stops, so that's why it thinks there are no PDO maps. As a work-around you could manually create the 1st PDO from the 2nd PDO (copy 1801 and 1A01 to 1800 and 1A00).

christiansandberg, 2017-02-07:
I made a change which hopefully fixes it. You can test the latest master and re-open the issue if it still doesn't work.

closed 2017-02-07 by christiansandberg

leongzhang, 2017-02-07:
thanks Chris, very helpful.
good day
gzhang

commit 6b587043 (2017-02-07): Support scattered PDO maps

Fixes #13