canopen-python/canopen #14: SDO block transfer

obbardc, 2017-02-08:
Hi,
This is an excellent looking library! I wish I knew about this a while ago.

I have the (thrilling) task of rewriting some CAN master software I wrote in C# on Windows to Raspberry Pi running Linux, this project looks perfect.

The product I am writing the software for has a bootloader which is written through SDO block transfer.

So I was wondering whether this can be added into the library?

Cheers!

christiansandberg, 2017-02-08:
I'm glad that you find it useful! Block transfer is definitely on the list, but I don't have a node which supports it to test it against. Right now I'm relying on someone else to give it a go.

obbardc, 2017-04-29:
Hi Christian,

I have a (custom-made) device that supports block-transfer for uploading firmware. Let me know if this is any help :).

Cheers!

christiansandberg, 2017-04-29:
I can start an implementation for block download if you can test it. Maybe you could also help in fixing it if you feel up for it.

obbardc, 2017-04-29:
Sure, I wouldn't really know where to start with writing it from scratch,
I'm a bit of a beginner in canopen.

christiansandberg, 2017-10-09:
I've merged the untested block transfer implementation. Hopefully someone can test it and report any bugs as new issues.

closed 2017-10-09 by christiansandberg