canopen-python/canopen #16: 402 device profile

luminize, 2017-02-18:
Hi @christiansandberg 
First, thank you for sharing this repo. I hope it makes life a bit easier.

I've been using python-can in a machine setup for approx a year, and i've hacked my way around getting my devices into different modes.

I'm using CANOpen motors with a CiA 402 profile (drioves and motion control). These motors are ultimately put in step/dir mode so that the Machinekit stack can drive the motors.

Since you've already laid down a framework I'm looking if extending the canopen library with functionality for writing the  [Power State engine](http://en.nanotec.com/products/manual/N5_CAN_EN/general%252Fds402_power_state_machine.html/) of these profiles.

Would this be something that fits this library?

a side question:
What would be the best way for me to upload and download SDO messages without an Object Dictionary. I don't have the bytearray data, but instead the index, subindex and value.
for example, I would like to set a value to 0x6041, subindex 0, value 0x06. I'd like to reuse code to convert these to the bytearray which I can then use like nw.nodes[3].sdo.download(0x6041,0,0x06)

Cheers

christiansandberg, 2017-02-19:
Implementing specific device profiles here is a good idea. But of course the solution needs to be modular, so I think subclassing `canopen.Node` would be the most logical way to add functionality. Maybe add a subfolder named profiles or something and add modules for each profile.

You can use the [struct](https://docs.python.org/3/library/struct.html) library in Python to encode your value to bytes. For example if you want to download an UNSIGNED16 value, write `data = struct.pack('<H', 0x06)` and then use the `node.sdo.download` method like you did above. Similarly you can decode data using `struct.unpack()`.

luminize, 2017-02-19:
thanks, I'll look into that.

luminize, 2017-02-21:
@christiansandberg your library works like a charm! Life is definately a bit easier now.

could you think along with me about the best way to listen to changes of a specific register?

I can subclass a NodeScanner, but upon working with this I realise that the statusword of the 402 profile Power State (register 0x6041) is normally transmitted thru a TPDO (for example when a homing routine has finished) and not necessarily (but also happening) a SDO reply.

Would it be wise to extend a call back function to a change of the specific register in the OD?
Otherwise I'll be polling the OD for each message. Or would there be a better way?

luminize, 2017-02-21:
reading this https://github.com/christiansandberg/canopen/issues/8#issuecomment-273157740 I see that you have a callback in pdo's. I'll look into that for sure.

christiansandberg, 2017-04-16:
Hi! How is the homing functionality going? I'm thinking I should release 0.5.0 soon so if you want it released together with the rest of the 402 profile then I can wait.

luminize, 2017-04-16:
> On 16 Apr 2017, at 21:24, Christian Sandberg wrote:
> 
> Hi! How is the homing functionality going? I'm thinking I should release 0.5.0 soon so if you want it released together with the rest of the 402 profile then I can wait.

I've got it working so it shouldn't be much work to make a PR. Need to add some additions to the docs. Been a bit busy so it slipped my mind to make a PR. I'll try to find some time these days to do that.

christiansandberg, 2017-08-12:
Hello @luminize, I will make a new attempt to make a release in the coming weeks if you want any changes to make it to 0.5.0.

luminize, 2017-08-14:
@christiansandberg I'll get the latest (tiny) additions from another PC and I'll make a PR. Short: I've added homing functionality and a callback can be added on the state change of the 402 state machine.

af-silva, 2019-10-16:
Hi @luminize,

Since this issue some changes happened to the canopen API for the DS402 profile.
I hope to hear from you regarding some of the changes I have made based on your initial work, any suggestions are welcome. I will close this Issue to clean up some of the threads opened but if you feel the need to reopen this or create a new one feel free to do so. 
Thanks for your contribution to this project.
Best regards ;)

closed 2019-10-16 by af-silva

luminize, 2019-10-16:
Hi @af-silva thanks, I have seen your work and it looks great. You're right to close this issue.
Best, Bas