canopen-python/canopen #17: Novice question

sqwang0, 2017-02-20:
Hi:
I am student who is learning CANopen
Now i am trying to  implement your code on raspberry and controlling an electronic car.
The system goes well when i sent NMT and SYNC command
but when a try to Read a variable using SDO
I have put the eds document into folder named objectdictionary
I received error like that
Could please advice me what mistake i might make.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/canopen-master/canopen/sdo.py", line 155, in __getitem__
    entry = self.od[index]
  File "/home/pi/canopen-master/canopen/objectdictionary/__init__.py", line 66, in __getitem__
    return self.names.get(index) or self.indices[index]
KeyError: 8197

Regards,

Siqi

christiansandberg, 2017-02-20:
Have you specified the EDS file when you created your node?
`network.add_node(6, '/path/to/object_dictionary.eds')`

Does the EDS file have index 0x2005 in it?

sqwang0, 2017-02-21:
Hi christian:

Yes i did. I have attached the error document and eds discription files.

There is 0x2005 exist. why?

I do appreciate your help, thanks!

Regards,

SIQI

University of Tasmania Electronic Communications Policy (December, 2014).
This email is confidential, and is for the intended recipient only. Access, disclosure, copying, distribution, or reliance on any of it by anyone outside the intended recipient organisation is prohibited and may be a criminal offence. Please delete if obtained in error and email confirmation to the sender. The views expressed in this email are not necessarily the views of the University of Tasmania, unless clearly intended otherwise.

pi@raspberrypi:~/canopen-master $ sudo /sbin/ip link set can0 up type can bitrate 250000
pi@raspberrypi:~/canopen-master $ python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11)
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
KeyError: 5120
KeyError: 5120

christiansandberg, 2017-02-21:
The first issue is that you have specified an absolute path /objectdictionary/port08step_byte.eds which does not exist. The second issue is that this should have raised an exception but it didn't which should be fixed now.

Try the latest development version:
`pip install --upgrade --pre canopen`

christiansandberg, 2017-02-25:
Please reopen this if changing the path to the EDS-file doesn't work.

closed 2017-02-25 by christiansandberg