canopen-python/canopen #18: NodeScanner usage

staircaseblog, 2017-02-21:
Thanks for the nice library. I looks clean and organized. I tend to migrate project which makes use of libcanopen to your library and make my first steps.

My first step is a node scanner. However it seems i am doing something wrong in the lines below. The return is allways empty []

Thanks in advance for pointers and best from south germany. BTW can_logger.py is able to listen.

import canopen
import time

network = canopen.Network()
network.connect(channel='can0', bustype='socketcan')
node = network.add_node(6, 'sample.eds')

scanner = canopen.network.NodeScanner(network)
time.sleep(30)
print scanner.nodes

luminize, 2017-02-21:
There's already a nodescanner in the network.
`network.listeners` gives you a list of the Listeners, normally these are `MessageListener` and `NodeScanner` objects.
```python
can0.listeners
[<canopen.network.MessageListener object at 0x7fb0223083d0>, <canopen.network.NodeScanner object at 0x7fb022308390>]
```
If you want to add another listener (to do custom stuff or something) to your network so it can receive callbacks you can add it to the listteners in your network.
```python
network.listeners.append(scanner)
```
but in your case it's already there :)

christiansandberg, 2017-02-21:
I've added a short example in the [documentation](http://canopen.readthedocs.io/en/latest/network.html).

staircaseblog, 2017-02-21:
Thanks. That was exactly what i was looking for.

closed 2017-02-22 by christiansandberg