canopen-python/canopen #19: How to send PDO3_RX with payload?

staircaseblog, 2017-02-21:
I need  to get a bit more familiar with canopen API to translate my old code. I need to run a PDO3_RX against a node "to_node_id" and i am wondering how i am going to do this with canopen and python. How can i send a PDO3_RX with payload is my first question. The second question is how i register a handler to listen for a response i am expecting.

Below is how the PDO3 request was build in my old code. Any help is highly appreciated. Thanks and kind regards

frame->function_code = CANOPEN_FC_PDO3_RX; //0x08 << 7 = 0x400
    frame->rtr = 0;

    frame->data_len = 8;
    frame->payload.data[0] = 0x80 | (to_node_id & 0x7F); //SDO request with server node_id
    frame->payload.data[1] = action; //00
    frame->payload.data[2] = 0x1f; //1F
    frame->payload.data[3] = 0x00; //subindex 0x00
    frame->payload.data[4] = 0x00 | (to_node_id & 0x7F); //server
    frame->payload.data[5] = 0x00 | (from_node_id & 0x7F); //client
    frame->payload.data[6] = 0x80;
    frame->payload.data[7] = 0x12;

    canopen_frame_send(sock, frame);
    canopen_frame_free(frame);

christiansandberg, 2017-02-22:
I assume this is some kind of proprietary peer-to-peer protocol using CAN-ID 0x400 to not clash with regular PDOs (since node ID 0 does not exist)? In that case I think it's best if you just send raw CAN frames.

```python
payload = bytearray(8)
payload[0] = 0x80 | (to_node_id & 0x7F) # SDO request with server node_id
...
network.send_message(0x400, payload)
```

You can listen for a specific CAN-ID using [Network.subscribe()](http://canopen.readthedocs.io/en/latest/network.html#canopen.Network.subscribe).

```python
def p2p_receive(can_id, data, timestamp):
    print('Got data %r' % data)

network.subscribe(0x380, p2p_receive)
```

staircaseblog, 2017-02-23:
Great thanks,

 i got the first step done. On the wire i see my request and response (btw. not the 0x400 as can_id but the 0x400 | from_node_id). However it looks i am doing something wrong registering the listeners. I am expecting the callback to fire when seeing  0x401 as can_id on the wire.  I do see the response on the wire, but the callback stays quiet.

Any idea? Thanks in advance and kind regards

`import canopen
import time

from_node_id = 44

def vA_Handler(can_id, data, timestamp):
    print('From %d got data %r' % (can_id,data))

def registriere_pdo_callbacks(nodes):
    for node in nodes:
        print('register PDO callback for node %d' % node)
        network.subscribe( 0x400|node, vA_Handler)

def get_cobid(to_node_id):
    AKTIVIERUNG = 0x00
    DEAKTIVIERUNG = 0x01
    global from_node_id

    print('get_cobid: i am %d, get cob-id with node %d (PDO3_RX --> %x)' % (from_node_id, to_node_id, 0x400| from_node_id))
    print('   expecting response from node %d with address %x' % (to_node_id, 0x400| to_node_id))

    # This happens on the wire
    # request from 44 to 1
    #  can0  42C   [8]  81 00 1F 00 01 2C 80 12
    # response back
    #  can0  401   [8]  AC 80 12 01 6C 06 00 00

    payload = bytearray(8)
    payload[0] = 0x80 | (to_node_id & 0x7F)
    payload[1] = AKTIVIERUNG
    payload[2] = 0x1f
    payload[3] = 0x00
    payload[4] = 0x00 | (to_node_id & 0x7F)  # server node id
    payload[5] = 0x00 | (from_node_id & 0x7F)  # client node id
    payload[6] = 0x80
    payload[7] = 0x12

    network.send_message((0x400 | from_node_id), payload)

network = canopen.Network()
network.connect(channel='can0', bustype='socketcan')
node = network.add_node(6, 'sample.eds')

network.scanner.search()
time.sleep(10)

print network.scanner.nodes
registriere_pdo_callbacks(network.scanner.nodes)
time.sleep(3)
get_cobid(1)

 `

christiansandberg, 2017-02-23:
You may need to sleep some more after `get_cobid(1)`, otherwise the program will exit right after transmitting 0x42C.

staircaseblog, 2017-02-23:
;-) oh that was a rookie fault. i owe you a drink!

closed 2017-02-23 by staircaseblog