canopen-python/canopen #20: SDO-Server

sasa00, 2017-02-22:
Hello,

Is it possible to use the canopen library as a sdo server? I'm trying to build a simulator for a canopen hardware and as far as I've seen it's only possible to use as sdo client.

Kind regards,
Sascha

christiansandberg, 2017-02-22:
No, there is no SDO server at the moment and I don't have any plans to implement that. But if you want to implement it then I would accept a pull request!

TobiasPleyer, 2017-10-27:
Hi Christian,

I would also have an interest in a SDO server implementation for the project I'm currently working on. I am curious about the reasons that kept you from adding this to the library. Are there architectural hurdles or is it a time cost problem? If you don't have the time to do all the work by yourself I would be happy to support you. I think SDO server support covers a big area of interest, e.g. device mocking in certain test scenarios.

Cheers,
Tobias

christiansandberg, 2017-10-27:
It’s more due to time constraint and that I myself did not need it. But if anyone is interested in implementing it I would happily accept it. We just need to figure out the architecture first. We will probably need a slave node class first.

TobiasPleyer, 2017-10-27:
Nice to hear

I would love to come to a solution with your help. Shall we have the discussion here in the chat or is it more appropriate to do this via another channel?
A small side question: It is maybe possible for me to make this change sponsored by my employer. How big would you estimate the time cost for a full time developer? Just a coarse guess.

Cheers

christiansandberg, 2017-10-29:
We can continue the discussion in #47.

af-silva, 2019-10-16:
As mention by Christian, the discussion regarding this issue moved to #47 . Follow up there, closing this issue.

closed 2019-10-16 by af-silva