canopen-python/canopen #21: sdo upload

staircaseblog, 2017-02-23:
Hi again,

the last thing i need is to find the equivalent to this expedited sdo upload.

canopen_sdo_upload_exp(sock, cob_id, 0x1018, 0x01, &vendor);

My translation of the call above is this below. However i am not sure if it is a good idea to instantiate add_node with a cob_id. The flow i need to follow to is to get a cob_id with to_node_id, then doing the sdo uploads against (cob_id, index, subindex) and then deregister the cob_id again. I am not using the dictionary here. I just have it in to avoid complaints.

	node = network.add_node(cob_id, 'sample.eds')
	vendor_id = node.sdo[0x1018][0x01]

	print vendor_id.raw

I can observe it is repeating in doing the SDO (looks correct to be) and getting a response back from.
cob-id: 6C, from_node_id: 2C, to_node_id: 0x01

  can0  66C   [8]  40 18 10 01 00 00 00 00
  can0  5EC   [8]  43 18 10 01 CB 00 00 00
  can0  66C   [8]  40 18 10 01 00 00 00 00
  can0  5EC   [8]  43 18 10 01 CB 00 00 00
  can0  66C   [8]  40 18 10 01 00 00 00 00

It looks like the filter for the response does not match what is on the wire. Below is the thrown error. 
Thanks and kind regards again.

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/can/notifier.py", line 42, in rx_thread
    callback(msg)
  File "/usr/local/lib/python2.7/dist-packages/can/CAN.py", line 47, in __call__
    return self.on_message_received(msg)
  File "/usr/local/lib/python2.7/dist-packages/canopen/network.py", line 206, in on_message_received
    self.network.notify(msg.arbitration_id, msg.data, msg.timestamp)
  File "/usr/local/lib/python2.7/dist-packages/canopen/network.py", line 171, in notify
    callback(can_id, data, timestamp)
  File "scan_canbus.py", line 14, in check_model
    print vendor_id.raw
  File "/usr/local/lib/python2.7/dist-packages/canopen/common.py", line 64, in raw
    value = self.od.decode_raw(self.data)
  File "/usr/local/lib/python2.7/dist-packages/canopen/common.py", line 24, in data
    return self.get_data()
  File "/usr/local/lib/python2.7/dist-packages/canopen/sdo.py", line 219, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "/usr/local/lib/python2.7/dist-packages/canopen/sdo.py", line 94, in upload
    return ReadableStream(self, index, subindex).read()
  File "/usr/local/lib/python2.7/dist-packages/canopen/sdo.py", line 331, in __init__
    response = sdo_client.send_request(request)
  File "/usr/local/lib/python2.7/dist-packages/canopen/sdo.py", line 70, in send_request
    raise SdoCommunicationError("No SDO response received")
SdoCommunicationError: No SDO response received

christiansandberg, 2017-02-23:
The problem is that you are doing the SDO reading directly in your callback. The callback is called from the same thread that receives CAN messages, so after the SDO request has been transmitted, the thread blocks waiting for the response which never comes since it also blocked incoming messages.

You will need to make the SDO reading in your main thread instead, perhaps using some kind of synchronization mechanism like a [threading.Event](https://docs.python.org/2.7/library/threading.html#event-objects) to wait for check_model() to be called.

staircaseblog, 2017-02-23:
Can SDO API calling happen only from the main thread or from any other thread different from the listener?

christiansandberg, 2017-02-23:
Any other thread is fine.

staircaseblog, 2017-02-24:
Yep. That works. So instead using the threading.Event here i parameterize a callback handler function which spawns a thread. Will change the handler to enqueue callback function and parameters for consumption by a singleton worker thread. 

	import canopen
	import time
	import threading
	import logging

	def checkUVR_tf(can_id, data, timestamp):
		cob_id = data[4] 

		if cob_id != 0:
		    node = network.add_node(cob_id, 'sample.eds')
		    vendor = node.sdo[0x1018][0x01]
		    product = node.sdo[0x1018][0x02]

		    if vendor.raw == 0xCB and product.raw == 0x100B:
		        print('UVR1611')

		    destroy_cobid(can_id)

	def create_cobid(to_node_id, fn):
		# define a function spawning a thread and handing in the call paramenteres can_id, data, timestamp
		def handler(c, d, t):
		    t1 = threading.Thread(name='helper', target=fn, args=(c,d,t))
		    t1.start()

		network.subscribe(0x400| to_node_id, handler)
		cob_pdo(to_node_id, create=True)

	create_cobid(1, checkUVR_tf)

closed 2017-02-24 by staircaseblog