canopen-python/canopen #22: Accessing objects not defined in EDS file

staircaseblog, 2017-02-24:
Hi again,

i am doing another sdo upload to an index which is not defined in the EDS file.

tag = node.sdo[0x2014][0x01]

The thrown exception looks like a failed lookup in the dictionary. I was able to create partly valid entries into the EDS but got  back:

Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "scan_canbus.py", line 34, in getUVRtime_tf
    print('%d-%d-%d' %(tag.raw[0], monat.raw[0], jahr.raw[0]))
  File "/usr/local/lib/python2.7/dist-packages/canopen/common.py", line 64, in raw
    value = self.od.decode_raw(self.data)
  File "/usr/local/lib/python2.7/dist-packages/canopen/objectdictionary/__init__.py", line 268, in decode_raw
    "Mismatch between expected and actual data size")
ObjectDictionaryError: Mismatch between expected and actual data size

Is the code my only friend for documentation of EDS?

Thanks in advance and best.

staircaseblog, 2017-02-24:
ok. i get it.

staircaseblog, 2017-02-24:
canopen/canopen/objectdictionary/datatypes.py is my friend

closed 2017-02-24 by staircaseblog

christiansandberg, 2017-02-24:
You can also use `node.sdo.upload(0x2014, 1)` and decode the returned bytes yourself.

http://canopen.readthedocs.io/en/latest/sdo.html#canopen.sdo.SdoClient.upload