canopen-python/canopen #24: sdo on windows

wodz, 2017-03-03:
Hi,
I am having hard time with current lib on windows. I tried 0.3 from pip and it worked (having known issues but still). Current git master (as of today) gives error when reading from device. Here is example session:
>>> import canopen
>>> n = canopen.Network()
>>> n.connect(bustype='ixxat', channel=0, bitrate=100000)
>>> node = n.add_node(125, 'my.eds')
>>> node.sdo[0x1000].raw
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "canopen\common.py", line 64, in raw
    value = self.od.decode_raw(self.data)
  File "canopen\common.py", line 24, in data
    return self.get_data()
  File "canopen\sdo.py", line 229, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "canopen\sdo.py", line 104, in upload
    return ReadableStream(self, index, subindex).read()
  File "canopen\sdo.py", line 342, in __init__
    response = sdo_client.send_request(request)
  File "canopen\sdo.py", line 78, in send_request
    raise SdoCommunicationError("No SDO response received")
canopen.sdo.SdoCommunicationError: No SDO response received

Other things I tried seems to work (Network.scanner() returns list of nodes, node.nmt.send_command() does send proper packet). The most annoying thing is that on linux with 'socketcan' driver exactly the same code with exactly the same EDS works. In both cases python-can and python-canopen are the same version.

Any hints welcome

christiansandberg, 2017-03-03:
Could be some timing issue with IXXAT, try increasing [the timeout](http://canopen.readthedocs.io/en/latest/sdo.html#canopen.sdo.SdoClient.RESPONSE_TIMEOUT). It was much higher in previous versions.

christiansandberg, 2017-03-03:
Or it could be the filter settings. Try disabling it with `n.connect(bustype='ixxat', channel=0, bitrate=100000, can_filters=None)`.

wodz, 2017-03-06:
can_filters=None did the trick. Thanks.

commit dac24511 (2017-03-06): Skip CAN filtering, it does not work for IXXAT

Fixes #24
closed 2017-03-06 by christiansandberg