canopen-python/canopen #25: strange problem with sdo transfers

wodz, 2017-03-08:
Here is my session with corresponding candump to see what is put on the wire:

l.network = canopen.Network()
l.network.connect(channel='can0', bustype='socketcan', bitrate=100000)
l..node = self.network.add_node(id, 'LPCBootrom.eds')

#unlock
l.node.sdo[0x5000].raw = 0x5a5a
#can0  67D   [8]  2B 00 50 00 5A 5A 00 00
#can0  5FD   [8]  60 00 50 00 00 00 00 00

#prepare sectors 0 7
l.node.sdo[0x5020].raw = 0x0700
#can0  67D   [8]  2B 20 50 00 00 07 00 00
#can0  5FD   [8]  60 20 50 00 00 00 00 00

#erase sector 0 7
l.node.sdo[0x5030].raw = 0x0700
#can0  67D   [8]  2B 30 50 00 00 07 00 00
#can0  67D   [8]  2B 30 50 00 00 07 00 00
#can0  5FD   [8]  60 30 50 00 00 00 00 00
#can0  5FD   [8]  80 30 50 00 09 00 00 0F

Now, objects 0x5020 and 0x5030 are defined exactly the same in EDS but in the second case transfer is performed twice. Why? The second transfer fails (with custom error code thanks to NXP) as write to 0x5030 must be preceded by write to 0x5020.

[LPCBootrom.eds.txt](https://github.com/christiansandberg/canopen/files/827648/LPCBootrom.eds.txt)

wodz, 2017-03-08:
I figured out it is device slow response. Setting  l.node.sdo.RESPONSE_TIMEOUT=0.3 solves the problem (which means the second transfer is actually retransmit)

christiansandberg, 2017-03-08:
I have struggled with setting a good default timeout time. If it's too long, the application will block unnecessarily long. If it's too short, things like this happen. I don't think the standard specifies this either:

> Time-out values are considered to be implementation specific and do not fall within the scope of this specification. However, it is recommended that an implementation provides facilities to adjust these time-out values to the requirements of the application.

I think I'll increase the timeout again and also skip retrying by default. It was good to have that on really crappy CAN buses but should probably not be the default behavior.

commit 980ff945 (2017-03-08): Change SDO timeout settings

Increase timeout to 300 ms
Do not retry by default
Closes #25.
closed 2017-03-08 by christiansandberg