canopen-python/canopen #30: [Q] Novice problem and question

MihaiLupoiu, 2017-05-22:
Hello @christiansandberg 

I have started recently to read about CanOpen and wanted to test. Which is when I found your repository.

I have been tried to read the heartbeat value, but unfortunately I am not capable of doing it..

The code is the next one (very similar to the example):
```
    # Start with creating a network representing one CAN bus
    network = canopen.Network()

    # Connect to the CAN bus
    network.connect(bustype='socketcan', channel='can0')

    # Add some nodes with corresponding Object Dictionaries
    node = network.add_node(7, '/home/mlupoiu/Documents/workspace/CANopenTest/bin/object_dictionarys/CANopenSocket.eds')

    node.nmt.state = 'OPERATIONAL'

    print node.object_dictionary[0x1017].name

    # Read a variable using SDO
    heartbeat = node.sdo[0x1017]
    print "The heartbeat is 0x%X" % heartbeat.raw

    # Write a variable using SDO
    #node.sdo['Producer heartbeat time'].raw = 1000

    # Disconnect from CAN bus
    network.disconnect()
```
I am capable of reading the heartbeat name, but wen I try to read the value i get the next error:
```
Producer heartbeat time
Traceback (most recent call last):
  File "./src/simple_example.py", line 57, in <module>
    main()
  File "./src/simple_example.py", line 27, in main
    print "The heartbeat is 0x%X" % heartbeat.raw
  File "/home/mlupoiu/Documents/workspace/CANopenTest/external_lib/canopen/canopen/common.py", line 64, in raw
    value = self.od.decode_raw(self.data)
  File "/home/mlupoiu/Documents/workspace/CANopenTest/external_lib/canopen/canopen/common.py", line 24, in data
    return self.get_data()
  File "/home/mlupoiu/Documents/workspace/CANopenTest/external_lib/canopen/canopen/sdo.py", line 199, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "/home/mlupoiu/Documents/workspace/CANopenTest/external_lib/canopen/canopen/sdo.py", line 109, in upload
    with ReadableStream(self, index, subindex) as fp:
  File "/home/mlupoiu/Documents/workspace/CANopenTest/external_lib/canopen/canopen/sdo.py", line 311, in __init__
    response = sdo_client.send_request(request)
  File "/home/mlupoiu/Documents/workspace/CANopenTest/external_lib/canopen/canopen/sdo.py", line 83, in send_request
    raise SdoCommunicationError("No SDO response received")
canopen.sdo.SdoCommunicationError: No SDO response received
```
And with candump I see the Bootup message and the SDO request, but I don't receive any response. I don't know if I am missing something... 

And if i want to communicate with other nodes that are not created in python? I noticed that with node.scanner we can scan connected nodes to the network, but is it posible to send SDO messages  to them?
For example I have used the [CANopenSocket](https://github.com/CANopenNode/CANopenSocket) to create a node, but  did not saw any mechanism to send a message to it without using send_message(can_id, data, remote=False) which is a raw CAN message.

Thank you in advance!

Best regards,
Mihai A. Lupoiu

christiansandberg, 2017-05-22:
Is the node you're trying to communicate with created with CANopenSocket? Does it have node ID 7? What is the CAN ID of the boot-up message?

Maybe you have misunderstood this library. It does not create any nodes, it only communicates with existing ones. You can for example use the scanner to find the node ID's present on the bus, then use ``network.add_node(node_id, '/path/to/od.eds')`` to add them and then you can read and write using SDO for example.

MihaiLupoiu, 2017-05-22:
Hi @christiansandberg 

You are wright. I had misunderstood the library, my bad. I thought I was creating the nodes, not linking to them. The CANopenSocket had the id 4 which I was able to see but I assigned it the node 7. 
I just tested it with id 4 in both CANopenSocket and in my python scripy and everything worked perfectly.

Thank you very much and sorry for the dumb question! :)

closed 2017-05-22 by MihaiLupoiu