canopen-python/canopen #32: Value does not fit in specified type error

luminize, 2017-05-22:
Hi,

I need to set a value 0xFF in register 0x6060 by means of an SDO.

my eds shows:
```INI
[6060]
ParameterName=Modes of operation
ObjectType=0x7
DataType=0x0002
AccessType=rww
DefaultValue=0x00
PDOMapping=1
```
I use the following which works fine for the homing mode (0x06):
```python
self.node.sdo['Modes of operation'].raw = 0x06
```

I get the following error:
```python
Traceback (most recent call last):
  File "<console>", line 1, in <module>
  File "/home/machinekit/canopen/canopen/common.py", line 78, in raw
    self.data = self.od.encode_raw(value)
  File "/home/machinekit/canopen/canopen/objectdictionary/__init__.py", line 320, in encode_raw
    raise ValueError("Value does not fit in specified type")
ValueError: Value does not fit in specified type
```

Any thought on how can I solve this to that I can send this value via SDO?

luminize, 2017-05-22:
Seems I just solved it by using `.phys`
```python
m.joints[20].node.sdo['Modes of operation'].phys = -1
```
which sends:
```
  can0  614   [8]  2F 60 60 00 FF 00 00 00
  can0  594   [8]  60 60 60 00 00 00 00 00
```

Any reason why I cannot send `0xFF` thru `.raw` ?

closed 2017-05-22 by luminize

christiansandberg, 2017-05-22:
It's because the data type is INTEGER8 which has a valid range of -128 to 127, so 255 is not valid.

luminize, 2017-05-22:
Thanks, understood