canopen-python/canopen #33: Unable to configure PDO "KeyError: 1"

TinusScheppers, 2017-05-23:
Hi,

We are trying to configure the PDO object of a Curtis motor controller.
However after executing:

`print(node.pdo.read())
node.pdo.tx[1].clear()
node.pdo.tx[1].add_variable(0x1018, 1)  # Vendor Id
node.pdo.tx[1].add_variable('Motor Speed')  # get RPM
node.pdo.tx[1].trans_type = 1
node.pdo.tx[1].enabled = True`

We receive:

`None
Traceback (most recent call last):
  File "driver.py", line 59, in <module>
    node.pdo.tx[1].clear()
  File "/usr/local/lib/python3.4/dist-packages/canopen/pdo.py", line 133, in __getitem__
    return self.maps[key]
KeyError: 1`

It seems that no config could be read, which might seem logical because there has not yet been any config made. However, how can we make the config when we cannot access any of the mapping?

Attached you can find the eds used:
[os123xes.txt](https://github.com/christiansandberg/canopen/files/1023214/os123xes.txt)

Thanks in advance!

Kind regards
Tinus

christiansandberg, 2017-05-24:
Strange. Can you try this minimal example (which works for me):

```python
import canopen
network = canopen.Network()
node = network.add_node(1, 'os123xes.eds')
node.pdo.tx[1].clear()
```

TinusScheppers, 2017-05-25:
Somehow a copy paste mistake sneaked in the node assignment causing it to not find the eds file.
Now it's working fine!

Case, closed, thanks!

closed 2017-05-25 by TinusScheppers

christiansandberg, 2017-05-25:
There is a bug where no exception is raised if the EDS-file does not exist. It was first discovered in issue #17 and fixed in the latest development releases.