canopen-python/canopen #34: CAN data from server not datatype of eds, results in type mismatch (or wrong data).

TinusScheppers, 2017-06-26:
Hi,

It seems that our CAN server (Curtis motor driver) does not reply with bytes in the datatype defined in the eds.

eg: motor_rpm is defined as Int16 in the eds, but we receive b'\xAA\xFF\x00'. This results in "Mismatch between expected and actual data size". If I change to a bigger int we get the unsigned value instead of the signed one (as expected).

How would I go about fixing this given that we cannot change what the CAN server sends, only what is defined in the eds and this drivers code?

I have also noticed that unpack cannot handle extra zero bytes (\x00), but unpack_from (without offset) ignores the extra (\x00) bytes and displays the value correctly.

```python
up = struct.unpack('h', b'\xAA\xFF')
print(up[0]) #successful

up1 = struct.unpack('h', b'\xAA\xFF\x00\x00')
print(up1[0]) #struct.error: unpack requires a bytes object of length 2

up2 = struct.unpack_from('h', b'\xAA\xFF\x00')
print(up2[0]) #successful
```

I will test using unpack_from instead of unpack in the canopen driver next week but I thought I would ask you for your opinion in the meanwhile.

Thanks in advance!

Kind regards
Tinus

christiansandberg, 2017-06-27:
If it's a bug in the CAN server and it is only on a limited number of variables, I would probably read the data and unpack it manually.

```python
data = node.sdo['Motor RPM'].data
motor_rpm, = struct.unpack_from('<h', data)
```

TinusScheppers, 2017-06-27:
This seems to be the easiest in the short run, requiring a bit of custom code for each datatype.

However, is it a design choice to use unpack instead of unpack_from? I understand that using unpack_from hides datatype errors, but given that you can't (always) control what the server sends, it might be better to also have a case where you would use unpack_from to avoid this error?

What is your opinion on this?

christiansandberg, 2017-06-27:
It's more that you shouldn't get weird side effects that are difficult to debug. If for example the EDS specifies INT16 while the server sends INT32. As long as the values are within INT16 range it will be fine, but when the values are bigger you will just get garbage values. It could suffice with a warning though but also that could be missed.

Also if you try to download values the server will likely not accept it either.

TinusScheppers, 2017-06-27:
Indeed, I will manually decode the data, which should suffice.

Thanks for the explanation!

acolomb, 2018-06-20:
Hi!

I'm also currently working on a project using Curtis Instruments motor controllers and came up with a couple of hacks and fixes to make python-canopen compatible with their (often broken) CANopen implementation.  My changes need to be separated out into single issues and documented, then I will make some PRs.

The issue at hand here bit me as well, and I worked around it partially by modifying the `ObjectDictionary` python object after import.  This way one can correct datatypes, and make raw <-> physical conversions work as well as defining units.

```python3
def curtis_1232e_dict():
    """CAN object dictionary template for the Curtis 1232E motor controller"""

    od = canopen.objectdictionary.import_od('../od/Curtis_1232E.eds')
    od['VCL_Throttle'].factor = 100.0 / 32767
    od['VCL_Throttle'].unit = '%'
    od['Motor_RPM'].factor = 1.0
    od['Motor_RPM'].unit = 'rpm'
    od['Motor_Temperature'].factor = 1.0 / 10
    od['Motor_Temperature'].unit = '°C'
    od['Analog1_Filtered'].factor = 10.0 / 1023
    od['Analog1_Filtered'].unit = 'V'
    od['VCL_PD_Throttle'].factor = 100.0 / 32766
    od['VCL_PD_Throttle'].unit = '%'
    od['user2'].data_type = canopen.objectdictionary.datatypes.INTEGER16
    od['user2'].factor = 360.0 / 8192
    od['user2'].unit = '%'
    od['BDI_Percentage'].data_type = canopen.objectdictionary.datatypes.UNSIGNED8
    od['BDI_Percentage'].factor = 1.0
    od['BDI_Percentage'].unit = '%'
    od['PD_Current'].factor = 2.0 / 607
    od['PD_Current'].unit = 'A'
    return od
```

Of course you could edit the EDS file as well... But since they are (crappily) generated by the Curtis software compiler, I didn't want to go that route, either.

There are some cases, though, where extra zero bytes are sent.  My primitive approach was to slice the byte array down to the needed size in `objectdictionary.Variable.decode_raw()`, similar to @TinusScheppers' approach using `unpack_from()`.  Can we somehow make this behaviour opt-in?  I suspect there are many CAN(open) devices that go the lazy route and always set DLC=8 for any message, regardless of actual content.

acolomb, 2018-06-26:
The extra data bytes issue is fixed on master now, Works For Me(tm) on the Curtis controllers.

@TinusScheppers if this issue is still relevant to you on its first anniversary, keep an eye on #86 for my other Curtis-related changes ;-)

christiansandberg, 2018-08-13:
Should work with v0.7.0 now.

closed 2018-08-13 by christiansandberg