canopen-python/canopen #37: Subindex not found when reading pdo

ChrisAlphabet, 2017-08-03:
Hi,

I am trying to familiarise myself with CANopen and came across your library (which is great!).

I am trying to read the PDO configurations but I am getting an error that states 'Subindex 5 was not found'. 

Are you able to explain what is happening when the PDO configuration is read? And what would the index be that is related to subindex 5? 

Thanks!

ChrisAlphabet, 2017-08-03:
Here is the error message aswell.

```
>>> akd_node.pdo.read()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/CAN-Servo-Control/canopen/canopen/pdo.py", line 61, in read
    pdo_map.read()
  File "/home/pi/CAN-Servo-Control/canopen/canopen/pdo.py", line 260, in read
    self.event_timer = self.com_record[5].raw
  File "/home/pi/CAN-Servo-Control/canopen/canopen/sdo.py", line 160, in __getitem__
    return Variable(self.sdo_node, self.od[subindex])
  File "/home/pi/CAN-Servo-Control/canopen/canopen/objectdictionary/__init__.py", line 117, in __getitem__
    raise KeyError("Subindex %s was not found" % subindex)
KeyError: 'Subindex 5 was not found'
>>> 
```

christiansandberg, 2017-08-03:
If the transmission type is 254 or 255, it reads subindex 5 which is the event timer. However, it should not raise an exception, just log that it could not read it and continue.

ChrisAlphabet, 2017-08-03:
Ok, cool. Would that depend on what was the default setting in the eds?

Also, I tried the changes that you just made and got a different subindex error, this time in `sdo.py`

```
Traceback (most recent call last):
  File "test_pdo.py", line 100, in <module>
    akd_node.pdo.read()
  File "/home/pi/CAN-Servo-Control/canopen/canopen/pdo.py", line 61, in read
    pdo_map.read()
  File "/home/pi/CAN-Servo-Control/canopen/canopen/pdo.py", line 271, in read
    value = entry.raw
  File "/home/pi/CAN-Servo-Control/canopen/canopen/common.py", line 64, in raw
    value = self.od.decode_raw(self.data)
  File "/home/pi/CAN-Servo-Control/canopen/canopen/common.py", line 24, in data
    return self.get_data()
  File "/home/pi/CAN-Servo-Control/canopen/canopen/sdo.py", line 199, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "/home/pi/CAN-Servo-Control/canopen/canopen/sdo.py", line 109, in upload
    with ReadableStream(self, index, subindex) as fp:
  File "/home/pi/CAN-Servo-Control/canopen/canopen/sdo.py", line 311, in __init__
    response = sdo_client.send_request(request)
  File "/home/pi/CAN-Servo-Control/canopen/canopen/sdo.py", line 89, in send_request
    raise SdoAbortedError(abort_code)
canopen.sdo.SdoAbortedError: Code 0x06090011, Subindex does not exist
```

christiansandberg, 2017-08-03:
It tries to read the map arrays (0x14XX and 0x18XX) by first reading subindex 0 to get the length of the array, then each subindex that should be present. For some reason one or more of those does not exist even if subindex 0 says it does.

Try [enabling some debug output](https://github.com/christiansandberg/canopen/blob/master/README.rst#debugging) and check which subindex goes wrong.

ChrisAlphabet, 2017-08-08:
I already had the logging enabled but it doesn't seem to do anything for me. Do I have to do anything other than adding the below?

```python
import logging
logging.getLogger('canopen').setLevel(logging.DEBUG)
logging.getLogger('can').setLevel(logging.DEBUG)
```

But I added some good old prints and was able to see that it is failing when it tries to access 0x1600 sub 2. Having a look at the `.eds` (available [here](http://www.kollmorgen.com/en-us/products/drives/servo/akd/) it looks like the default value is 0x01 so I am not sure why it is trying to access sub 2.

Any help would be greatly appreciated!

christiansandberg, 2017-08-08:
Now I see. I assumed that the maps were defined as arrays in the Object Dictionary, but the standard says it should be records. I just need to change the way we iterate over the map entries...

closed 2017-08-08 by christiansandberg

ChrisAlphabet, 2017-08-09:
Working perfectly now! Thanks! :smiley:

commit 950a3ef0 (2017-08-03): Do not fail if EventTimer is not in OD

Fixes #37
commit 546af4d4 (2017-08-08): Make PDO maping work regardless of if it is an array or record

Fixes #37