canopen-python/canopen #39: Difference between PDO and SDO reading of the same index

ChrisAlphabet, 2017-08-09:
Hello,

First off, thanks for the speedy `pdo.read` fix! Once all that was working I came across another issue. 

I am having an issue where if I read the value at an index using the sdo I get the expected value but when I try to use synchronous pdo transfers of the same index I get the wrong value.

Do pdo's do anything internally that might effect scaling?

I checked to make sure that no scaling factors have been introduced but I figured if a scaling factor was present it would've scaled the sdo aswell.

Any help is greatly appreciated!

Cheers

christiansandberg, 2017-08-09:
There should not be any difference. Both SDO and PDO use the same code for converting data to values. What does your code look like?

ChrisAlphabet, 2017-08-09:
```python
import canopen
import time
network = canopen.Network()
akd_node = network.add_node(1, 'AKD_CANopen.eds')
network.connect(bustype='socketcan', channel='can1')
akd_node.pdo.read()
akd_node.pdo.tx[1].clear()
akd_node.pdo.tx[1].add_variable('Analog output', 'AOUT_INPUT_VALUE')
akd_node.pdo.tx[1].trans_type = 10
akd_node.pdo.tx[1].event_timer = 0
akd_node.pdo.tx[1].enabled = True
akd_node.nmt.state = 'PRE-OPERATIONAL'
akd_node.pdo.save()
ain = akd_node.sdo['Analog output']['AOUT_INPUT_VALUE'].raw
print("AIN: ", ain/1000)
network.sync.start(0.1)
akd_node.nmt.state = 'OPERATIONAL'
try:
    while True:
        akd_node.pdo.tx[1].wait_for_reception()
        analog_value = (akd_node.pdo.tx[1]['Analog output.AOUT_INPUT_VALUE'].phys)
        print("Value: ", analog_value/1000)
        time.sleep(1)
        
except KeyboardInterrupt:
    print('Exiting while loop...')

akd_node.pdo.read()
akd_node.pdo.tx[1].clear()
akd_node.pdo.tx[1].enabled = False
akd_node.pdo.save()        
network.sync.stop()
network.disconnect()
```

ChrisAlphabet, 2017-08-09:
Having a look at the candump from can1 it looks like the pdo values that are being sent are the incorrect values, all of them off by a factor of 2.61. That suggests to me that something inside the akd is scaling the pdo.

ChrisAlphabet, 2017-08-10:
Looks very likely that this is an internal bug to the device I am using, not your library. I hand-carved the can messages using `python-can` and the incorrect pdo value popped up again. 

Hand-carving the can messages gave me an even greater appreciation for your library because it was quite a pain! :smile: 

Thanks again!

closed 2017-08-10 by ChrisAlphabet