canopen-python/canopen #42: Error cleaner TPDO

cleitonbueno, 2017-09-02:
I'm trying to implement a TPDO2 that will be called periodically every 5s and send a value, but when I start pdo.tx [2] I get this error:

```
# python commTest.py 
Creating object CanOpen
Read .eds and create node
Connecting socketCan with CanOpen
Init TPDO2

Traceback (most recent call last):
  File "commTest.py", line 25, in <module>
    node.pdo.tx[2].clear()
  File "/usr/lib/python2.7/site-packages/canopen/pdo.py", line 135, in __getitem__
    return self.maps[key]
KeyError: 2
```

Code:
```
import canopen
import sys
import struct

print("Creating object CanOpen")
network = canopen.Network()

print("Read .eds and create node")
node = network.add_node(1, 'model.eds')

print("Connecting socketCan with CanOpen")
network.connect(bustype='socketcan', channel='can0')

print("Init TPDO2")
node.pdo.read()
node.pdo.tx[2].clear()
```

The idea is a TPDO2 that the function will be executed every 5s, based on COB-ID 0x283

canopen version 0.5.0.dev5(db5f1d41b97313b7142117142d3d4b6975665537)

christiansandberg, 2017-09-03:
Could you send the EDS or the just the part with PDOs (0x1400-0xAFFF)? It doesn't seem to have found the maps for some reason.

af-silva, 2019-10-16:
No response to Christian question. Closing the issue.
@cleitonbueno if you feel the need to pursue a solution for your problem you can open a new issue or I can reopen this one. 
Thanks ;)

closed 2019-10-16 by af-silva