canopen-python/canopen #44: canopen.sdo.SdoCommunicationError: No SDO response received

ZhVS, 2017-10-12:
Hi!
I'm trying to start with your canopen python library, and I even can't make examples from git work without errors. :(

My code is a copy-paste one from example:
```
import canopen

# Start with creating a network representing one CAN bus
network = canopen.Network()

# Add some nodes with corresponding Object Dictionaries
node = network.add_node(6, 'd:\\MyDocs\\Libs\\py\\_work\\canopen-master\\test\\sample.eds')
network.add_node(7, 'd:\\MyDocs\\Libs\\py\\_work\\canopen-master\\test\\sample.eds')

# Connect to the CAN bus
# Arguments are passed to python-can's can.interface.Bus() constructor
# (see https://python-can.readthedocs.io/en/latest/bus.html).
#network.connect(bustype='socketcan', channel='can0')
# network.connect(bustype='kvaser', channel=0, bitrate=250000)
# network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate=250000)
network.connect(bustype='virtual', channel=0, bitrate=250000, can_filters = None)
# network.connect(bustype='vector', app_name='CANalyzer', channel=0, bitrate=250000)
# network.connect(bustype='nican', channel='CAN0', bitrate=250000)

#network.scanner.search()

# Read a variable using SDO
device_name = node.sdo['Manufacturer device name'].raw
vendor_id = node.sdo[0x1018][1].raw

# Write a variable using SDO
node.sdo['Producer heartbeat time'].raw = 1000

# Read PDO configuration from node
node.pdo.read()
# Re-map TxPDO1
node.pdo.tx[1].clear()
node.pdo.tx[1].add_variable('Application Status', 'Status All')
node.pdo.tx[1].add_variable('Application Status', 'Actual Speed')
node.pdo.tx[1].trans_type = 254
node.pdo.tx[1].event_timer = 10
node.pdo.tx[1].enabled = True
# Save new PDO configuration to node
node.pdo.save()

# Transmit SYNC every 100 ms
network.sync.start(0.1)

# Change state to operational (NMT start)
node.nmt.state = 'OPERATIONAL'

# Read a value from TxPDO1
node.pdo.tx[1].wait_for_reception()
speed = node.pdo.tx[1]['Application Status.Actual Speed'].phys

# Disconnect from CAN bus
network.sync.stop()
network.disconnect()
```
eds file used in code I've got from canopen-master\test\
And here is what a interpreter prints to console after runnig script
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\Tools\WinPython-2.7.6.4\python-2.7.6\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 540, in runfile
    execfile(filename, namespace)
  File "D:/MyDocs/SDBCS/_Trash/CAN/can_first_steps.py", line 30, in <module>
    device_name = node.sdo['Manufacturer device name'].raw
  File "D:\Tools\WinPython-2.7.6.4\python-2.7.6\lib\site-packages\canopen\common.py", line 64, in raw
    value = self.od.decode_raw(self.data)
  File "D:\Tools\WinPython-2.7.6.4\python-2.7.6\lib\site-packages\canopen\common.py", line 24, in data
    return self.get_data()
  File "D:\Tools\WinPython-2.7.6.4\python-2.7.6\lib\site-packages\canopen\sdo.py", line 240, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "D:\Tools\WinPython-2.7.6.4\python-2.7.6\lib\site-packages\canopen\sdo.py", line 150, in upload
    with ReadableStream(self, index, subindex) as fp:
  File "D:\Tools\WinPython-2.7.6.4\python-2.7.6\lib\site-packages\canopen\sdo.py", line 369, in __init__
    response = sdo_client.request_response(request)
  File "D:\Tools\WinPython-2.7.6.4\python-2.7.6\lib\site-packages\canopen\sdo.py", line 118, in request_response
    return self.read_response()
  File "D:\Tools\WinPython-2.7.6.4\python-2.7.6\lib\site-packages\canopen\sdo.py", line 102, in read_response
    raise SdoCommunicationError("No SDO response received")
canopen.sdo.SdoCommunicationError: No SDO response received

```
The same errors I've got when used ixaat usb-to-can II module and changed a code to

`network.connect(bustype='ixaat', channel=0, bitrate=250000, can_filters = None)`

What am I doing wrong?

christiansandberg, 2017-10-12:
Using the virtual CAN interface will definitely not give a response. Does the node you're trying to communicate with have a node ID of 6? Can you communicate with it with some other application? You  may need to analyze the CAN traffic to check that the request is sent with correct CAN ID and if a response is sent.

ZhVS, 2017-10-13:
Ok, looks like I'm missing something fundamental...

> Does the node you're trying to communicate with have a node ID of 6?

No, but I thought that it was created after 

`node = network.add_node(6, 'd:\\MyDocs\\Libs\\py\\_work\\canopen-master\\test\\sample.eds')`

What canopen (lib) makes while executing this line?
Is this a real node in CAN network or eds-file is just a 'template' for more comfortable work with PDOs and SDOs of a real node, realized on some another device?

christiansandberg, 2017-10-13:
This library only acts as a master. It communicates with other already existing nodes. There have been other requests for supporting creating/simulating nodes (e.g. #20) but it will probably require completely different code base.

ZhVS, 2017-10-16:
christiansandberg, tnx, it explains a lot. But I still can't understand what's the point of importing eds file...

christiansandberg, 2017-10-16:
The EDS is needed to know which parameters are available and what data type they are among other things.

maddilety, 2018-08-13:
I am facing similar issue when I am trying to write the data using node.sdo
i.e.
  File "D:/Maddy_Docs/MD2_Projects/POC/Muttu_Code/PyCanOpenSampleTest.py", line 303, in ProfileInitialize
    coSdoWrite(objIndex, objSubIndex, dataBytes)
    File "D:/Maddy_Docs/MD2_Projects/POC/Muttu_Code/PyCanOpenSampleTest.py", line 94, in    coSdoWrite
    node.sdo[index][subindex].raw = data
TypeError: 'Variable' object does not support indexing
Please let me know your feedback.

Thanks
Maddy

closed 2020-04-19 by af-silva

srishtigusain11, 2021-11-23:
Hello,
I am getting this NO SDO response received error. Could you please help me with this, I did try suggestions mentioned above.

Traceback (most recent call last):
  File "C:/Users/gusains/Downloads/Test/CANOpen.py", line 47, in <module>
    print("The device type is 0x%X" % device_type.raw)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\variable.py", line 74, in raw
    value = self.od.decode_raw(self.data)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\variable.py", line 36, in data
    return self.get_data()
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\base.py", line 108, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 117, in upload
    fp = self.open(index, subindex, buffering=0)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 197, in open
    raw_stream = ReadableStream(self, index, subindex)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 243, in __init__
    response = sdo_client.request_response(request)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 85, in request_response
    return self.read_response()
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 69, in read_response
    raise SdoCommunicationError("No SDO response received")
canopen.sdo.exceptions.SdoCommunicationError: No SDO response received
Bus error: an error counter reached the 'heavy'/'warning' limit
```
try:
    # Start with creating a network representing one CAN bus
    network = canopen.Network()
    # CAN bus connect
    network.connect(channel='PCAN_USBBUS1', bustype='pcan', bitrate=125000)
    network.check()

    # Add some nodes with corresponding Object Dictionaries
    nodeId = 1
    file_path = os.path.join(os.path.dirname(__file__), 'sample.eds')
    # Add nodes to the network
    node = network.add_node(nodeId, file_path)

    for obj in node.object_dictionary.values():
        print('0x%X: %s' % (obj.index, obj.name))
        if isinstance(obj, canopen.objectdictionary.Record):
            for subobj in obj.values():
                print(' %d: %s' % (subobj.subindex, subobj.name))
    """
        for node_id in network:
        print(network[node_id])
    """
    # Simple variables
    # Read
    device_type = node.sdo[0x1000]
    print("The device type is 0x%X" % device_type.raw)

    # Write

    identity_obj = node.sdo[0x1018]
    node.sdo[0x1018].raw = 4000
    network.sync.start(0.1)
    print("Updated identity_obj is 0x%X:" % identity_obj.raw)
except Exception as e:
    exc_type, exc_obj, exc_tb = sys.exc_info()
    fname = os.path.split(exc_tb.tb_frame.f_code.co_filename)[1]
    print(exc_type, fname, exc_tb.tb_lineno)
    traceback.print_exc()
finally:
    print('going to exit... stopping...')
    if network:
        network.disconnect()
```