canopen-python/canopen #45: RPDO for multiple devices

DPastl, 2017-10-23:
Hello,

I apologize if this isn't the right place to ask this, as I'm not sure it counts as an "issue".

I'm attempting to synchronize to two Allied Motion motors via RPDO messages.  I've successfully managed to get them all to transmit PDO messages on sync broadcasts, but I can't even get a single one to respond to PDO messages (Which I think I'm sending using alliedmotor1.pdo.rx[4]['EN'].raw = 1, or enable motor).  I'm guessing that I'm doing something wrong, but I can't quite see what.

```
import canopen
import time

network = canopen.Network()

# network.connect(channel='PCAN_USBBUS1', bustype='pcan', bitrate=125000)
network.connect(bustype='socketcan', channel='can0', bitrate=125000)
# network.nmt.state = 'RESET'
# network.nmt.state = 'OPERATIONAL'
alliedmotor1 = network.add_node(0x01, 'AM_Drive.eds')
alliedmotor5 = network.add_node(0x05, 'AM_Drive.eds')

alliedmotor1.sdo['DCV'].raw = 1
# alliedmotor1.sdo['EN'].raw = 1

alliedmotor5.sdo['DCV'].raw = 1
alliedmotor5.sdo['EN'].raw = 1

#Motor 1 PDO setup
alliedmotor1.pdo.read()
alliedmotor1.pdo.tx[1].clear()
alliedmotor1.pdo.tx[1].add_variable('CV')
alliedmotor1.pdo.tx[1].add_variable('DCV')
alliedmotor1.pdo.tx[1].trans_type = 1
alliedmotor1.pdo.tx[1].enabled = True

alliedmotor1.pdo.rx[1].clear()
alliedmotor1.pdo.rx[1].add_variable('EN')
alliedmotor1.pdo.rx[1].enabled = True

alliedmotor5.pdo.read()
alliedmotor5.pdo.tx[1].clear()
alliedmotor5.pdo.tx[1].add_variable('CV')
alliedmotor5.pdo.tx[1].add_variable('DCV')
alliedmotor5.pdo.tx[1].trans_type = 1
alliedmotor5.pdo.tx[1].enabled = True

network.nmt.state = 'PRE-OPERATIONAL'
alliedmotor1.pdo.save()
alliedmotor5.pdo.save()

network.sync.start(0.1)
alliedmotor1.pdo.rx[4]['EN'].raw = 1
alliedmotor1.pdo.rx[1].start(0.1)
network.nmt.state = 'OPERATIONAL'

for i in range(10):
    alliedmotor5.pdo.tx[1].wait_for_reception()
    print alliedmotor1.pdo.tx[1]['CV'].raw, alliedmotor5.pdo.tx[1]['CV'].raw, \
        alliedmotor1.pdo.tx[1]['DCV'].raw, alliedmotor5.pdo.tx[1]['DCV'].raw

alliedmotor1.sdo['EN'].raw = 0
alliedmotor5.sdo['EN'].raw = 0

network.sync.stop()
network.disconnect()
```

christiansandberg, 2017-10-24:
It seems you’re setting EN in RxPDO4 but starts transmission on RxPDO1.

DPastl, 2017-10-24:
....Well that's a bit embarrassing....  That fixed the problem!

Another question, if you don't mind.  How would one setup the RxPDO to be processed on sync reception rather than based on time?  What I'd like to do is sent a PDO to set the speed of each motor, then have the sync message tell them all to process that simultaneously.

christiansandberg, 2017-10-26:
Not sure if it's possible to make the RxPDO's unpack with the sync message. But if transmission type exists in your device for RxPDO's also, try setting it to 1. You can set both nodes to listen for the same COB-ID though.

closed 2018-04-20 by DPastl