canopen-python/canopen #46: Why does Maps use a range value of 32?

TobiasPleyer, 2017-10-24:
Thanks for the framework and putting your time in!

**File**: pdo.py
**Class**: Maps
**Function**: init

Why `map_no in range(32)`? I would expect 127 as range value.

Kind regards

christiansandberg, 2017-10-25:
It should be 512 even. Don't know how many are reasonable but it won't affect parsing speed much anyway so I'll change it to 512.

TobiasPleyer, 2017-10-25:
Thanks for the quick response. Can you explain the numbers to me please. I looked at the specification and from what I understood e.g. the receive PDO parameters range from index 1400h to 147Fh, from which I concluded the index count of `147Fh - 1400h = 127`. Why then 512?
Thanks a lot

christiansandberg, 2017-10-25:
I have an older version of the standard (4.02 from 2002) where it specifies 1400h - 15FFh as RPDO parameters. If yours is newer then you are probably right (except it's 128 since 1400h is counted as number 1). I doubt anyone would use more than 128 anyway.

commit d1849afb (2017-11-27): Scan 128 PDO maps

Fixes #46
closed 2017-11-27 by christiansandberg