canopen-python/canopen #49: PDO COB-ID DefaultValue issue

jra-skov-dk, 2017-11-06:
Hi.

Thanks for this great python canopen implementation.
I am using the EDS import function of python canopen to autogenerate C code for a canopen stack.

I have encountered a small issue in this regard.
```

[1401sub1]
ParameterName=COB ID
ObjectType=0x7
DataType=0x0007
AccessType=rw
DefaultValue=$NODEID+0x300
PDOMapping=0
LowLimit=0x00000001
HighLimit=0xFFFFFFFF
```

For all PDO configurations the COB-ID Default value looks like above `DefaultValue=$NODEID+0x300` (generated by VectorEds)
But it seems the line `var.default = int(eds.get(section, "DefaultValue"), 0)` in eds.py cannot handle this `$NODEID+` syntax.

If I remove `$NODEID+` in the eds I get a warnings from VectorEds that this differs from standard value.

Any suggestions on how to solve this?

My suggestion:
1. Extract DefaultValue as a string
2. Evaluate for `$NODEID+` and remove it
3. Convert to int and use as it is done now.

What do you think? (I don't think the `$NODEID+` is used for anything in the master, It is just an indication of behavior shown in the EDS file)

jra-skov-dk, 2017-11-06:
Just did a test:

Changing from this:
`var.default = int(eds.get(section, "DefaultValue"), 0)`
to this:
`var.default = int(eds.get(section, "DefaultValue").replace('$NODEID+',''), 0)`

fixes everything for me, all other DefaultValues stay the same but COB-ID is loaded correctly now.
I will make a pull request and you can evaluate if this has unintended side effects.

christiansandberg, 2017-11-06:
This is a known limitation (see #1). $NODEID should be replaced with the actual node ID and then evaluated. So if the node ID is 6, $NODEID+0x300 should be 0x306. I prepared for this by feeding the node_id to import_eds(). Could you try to implement that instead?

jra-skov-dk, 2017-11-06:
Ok, I see what you mean.

I will implement that instead. (lets continue the talk in https://github.com/christiansandberg/canopen/pull/50)

commit 10a3e958 (2017-11-06): Merge pull request #50 from jra-skov-dk/fix-eds-cob-id-default-value

Fixed issue with loading eds DefaultValues from COB-ID's
Fixes #49.
closed 2017-11-06 by christiansandberg