canopen-python/canopen #53: LSS fastscan, next release

iurly, 2018-01-05:
Hi,
at a first glance it looks like LSS support is almost complete (at least in the master branch) with only fastscan missing. Is anyone having a look at it? What's the current plan for the next release?

Thank you!

ymkim92, 2018-01-06:
I implemented the LSS functionality for my own testing purpose. In my list, there was no support for fastscan. I am also welcome for you to implement it!

iurly, 2018-01-06:
Thank you guys for your answers - I somehow suspected that would be the kind of response I would get. :-)
I don't really have much time to do that (I had a quick look at the Fastscan implementation in CANOpenNode and that kinda freaked me out...), but I'll definitely contribute it back if I do!

ymkim92, 2018-08-29:
I added LSS fastscan and sent a pull request for that.

ololoshka2871, 2018-10-04:
I'm new to Canopen, now I'm developing a new device (sensor). I use beaglebone black board as CAN master.
While debugging LSS, I was discovered the following problem:
I am using python 2.7.3. 
Apparently the LssMaster.send_command() method is waiting for the 'bytearray' input, but 'str' is coming.
I used the following workaround. Now, my_fast_scan() works.

```python
def new__send_command(self, message):
    m = bytearray()
    m.extend(message)
    message = m

    response = None
    if not self.responses.empty():
        self.responses = queue.Queue()

    self.network.send_message(self.LSS_TX_COBID, message)

    if not bool(message[0] in ListMessageNeedResponse):
        return response

    # Wait for the slave to respond
    # TODO check if the response is LSS response message
    try:
        response = self.responses.get(
            block=True, timeout=self.RESPONSE_TIMEOUT)
    except queue.Empty:
        raise LssError("No LSS response received")
    
    r = bytearray()
    r.extend(response)
    return r

def my_fast_scan(lssMaster, put_config_state=True):
    old__send_command = lssMaster._LssMaster__send_command
    funcType = type(lssMaster._LssMaster__send_command)
    lssMaster._LssMaster__send_command = funcType(new__send_command, lssMaster, canopen.lss.LssMaster)
    res = lssMaster.fast_scan()
    lssMaster._LssMaster__send_command = old__send_command
    if not put_config_state:
        lssMaster.send_switch_state_global(lssMaster.WAITING_STATE)
    return res
```

ymkim92, 2018-10-24:
Hi ololoshka2871,

I made a fix here:
https://github.com/christiansandberg/canopen/pull/117

af-silva, 2019-10-16:
Solution found at #117

closed 2019-10-16 by af-silva