canopen-python/canopen #58: test_PDO error

sorki, 2018-02-03:
Getting a following error when running `python setup.py test`. This only fires when logging is set to DEBUG and `Variable.set_data` tries to log a message referencing `self.msg.cob_id` which is None.

```
======================================================================
ERROR: test_bit_mapping (test_pdo.TestPDO)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/rmarko/git/canopen/test/test_pdo.py", line 20, in test_bit_mapping
    map['INTEGER16 value'].raw = -3
  File "/home/rmarko/git/canopen/canopen/common.py", line 78, in raw
    self.data = self.od.encode_raw(value)
  File "/home/rmarko/git/canopen/canopen/common.py", line 30, in data
    self.set_data(data)
  File "/home/rmarko/git/canopen/canopen/pdo.py", line 437, in set_data
    self.name, binascii.hexlify(data), self.msg.cob_id)
  File "/usr/lib64/python2.7/logging/__init__.py", line 1155, in debug
    self._log(DEBUG, msg, args, **kwargs)
  File "/usr/lib64/python2.7/logging/__init__.py", line 1286, in _log
    self.handle(record)
  File "/usr/lib64/python2.7/logging/__init__.py", line 1296, in handle
    self.callHandlers(record)
  File "/usr/lib64/python2.7/logging/__init__.py", line 1336, in callHandlers
    hdlr.handle(record)
  File "/usr/lib64/python2.7/logging/__init__.py", line 759, in handle
    self.emit(record)
  File "/usr/lib/python2.7/site-packages/nose/plugins/logcapture.py", line 82, in emit
    self.buffer.append(self.format(record))
  File "/usr/lib64/python2.7/logging/__init__.py", line 734, in format
    return fmt.format(record)
  File "/usr/lib64/python2.7/logging/__init__.py", line 465, in format
    record.message = record.getMessage()
  File "/usr/lib64/python2.7/logging/__init__.py", line 329, in getMessage
    msg = msg % self.args
TypeError: %X format: a number is required, not NoneType
```

TobiasPleyer, 2018-02-06:
Hi sorki

In your stack tarce I can see the file _common.py_, which tells me that you are on the master branch. The problem is that the support for PDOs, client and server side, is currently under active development. I had a quick look at the implemenation of the current master branch. The **cob_id** is only set when the **read** method of the **Map** object is called, which in return happens when the **PdoNode.read** method is called. If this does not happen then the cob_id remains None. Since the PDO test case does not call the read method the error above happens. This behaviour has changed! I recommend you switching temporarily to the **local-nodes** branch. On this branch the test should also run successfully.

@christiansandberg What do you think about this bug issue?

christiansandberg, 2018-02-06:
I can do a temporary fix on the master branch but I think I made a better long-term fix on the local-nodes branch where cob_id gets a default value based on direction, map number and node ID.

dankamongmen, 2019-06-07:
@christiansandberg need an assist here? I'm developing a small app atop your module, and would like a clean unit test firmament underneath me, especially as this is my first Python in over ten years heh. https://github.com/dankamongmen/canscan

commit 484c22d9 (2019-06-11): Fix debug logging not working in some cases

Fixes #58.
closed 2019-06-11 by christiansandberg