canopen-python/canopen #61: Problem while getting VISIBLE_STRING SDOs

tnivon, 2018-02-12:
When getting VISIBLE_STRING SDO if this one is correctly ended by a \0 char but not fully filled with \0 decoding generate an error.

I suggest to add these few lines to correct the problem.

diff --git a/canopen/objectdictionary/__init__.py b/canopen/objectdictionary/__init__.py
index c05fe7c..ab57d45 100644
--- a/canopen/objectdictionary/__init__.py
+++ b/canopen/objectdictionary/__init__.py
@@ -279,8 +279,14 @@ class Variable(object):
 
     def decode_raw(self, data):
         if self.data_type == VISIBLE_STRING:
+            endOfString=data.find(0) 
+            if endOfString>0:
+                data=data[:endOfString]
             return data.decode("ascii")
         elif self.data_type == UNICODE_STRING:
+            endOfString=data.find(0) 
+            if endOfString>0:
+                data=data[:endOfString]
             # Is this correct?
             return data.decode("utf_16_le")
         elif self.data_type in self.STRUCT_TYPES:

christiansandberg, 2018-02-12:
Normally you shouldn't zero terminate strings when using SDO. After all, the length of the string is known.

> There is no 0h necessary to terminate the string.

What is the error you get? I guess we should be able to accept zero terminated strings so I'll strip any trailing zeroes.

christiansandberg, 2018-02-13:
Then there is definitely an error in your node implementation. A VISIBLE_STRING must only contain visible characters. Do you know if it specifies the size of the string when the transfer is started?

I’ll have to consider if this should be handled here or if quirky nodes should be read using the .data attribute.

christiansandberg, 2018-07-16:
Could you test with latest master? I've done a change that could have fixed this depending on if the node specifies the size of the string at the start of transmission or not.

christiansandberg, 2018-08-13:
I'm hoping compatibility with some implementations works better with v0.7.0. Please re-open otherwise.

commit 3da049f6 (2018-02-12): Accept zero terminated strings and unknown charachters

Fixes #61
closed 2018-08-13 by christiansandberg