canopen-python/canopen #64: PDO and SDO expected and actual data size

morganPigeron, 2018-03-16:
Hi , 
I'm currently working on some motor controller. 
I got this error when i try to communicate using the "manufacturer address" : 
"Mismatch between expected and actual data size"

But I always dump the can and I can see that the  node sends the information I request. 

Exemple SDO: 
#in my case 0x27d5 is the can address 
nodeAdr = node.sdo[0x27d5].raw
"Mismatch between expected and actual data size"

But in the Can dump I can see  "584 [8] 42 D5 27 00 01 00 00 00"
so it matches the actual address wich is 1 (in the device) .

Exemple PDO:
when I do a :
 node.pdo.read()
"Mismatch between expected and actual data size"
the can dump is : 

000 [2] 80 04
604 [8] 40 00 14 01 00 00 00 00
584 [8] 42 00 14 01 04 02 00 00
604 [8] 40 00 14 02 00 00 00 00
584 [8] 42 00 14 02 FE 00 00 00

Thanks for your help.

christiansandberg, 2018-03-16:
Sorry, it also says that the size is not specified so that is correct. Then I guess we must use the Object Dictionary to determine how many bytes to actually use.

morganPigeron, 2018-03-17:
Thanks for your quick and usefull reply, 

That right we needed to specify the type of data. This kind of code work : 
``` python 
id = int(0x1400)
COB_ID = node.sdo[id][1].data
TRANSMISSION_TYPE = node.sdo[id][2]

print struct.unpack_from('B',COB_ID)
print struct.unpack_from('L',TRANSMISSION_TYPE.data)
```
However I'm a little bit stuck with PDO because i can't use the exemple code. read() and save() use SDO : 

``` python 
node.pdo.read()
node.pdo.tx[4].clear()
node.pdo.tx[4].add_variable('Application Status', 'Status All')
node.pdo.tx[4].add_variable('Application Status', 'Actual Speed')
node.pdo.tx[4].trans_type = 254
node.pdo.tx[4].event_timer = 10
node.pdo.tx[4].enabled = True
node.pdo.save()
```
 So I set up the PDO like that :

``` python
#node.pdo.tx[4].event_timer = 10
EVENT_TIMER = node.sdo[tx4][5]
EVENT_TIMER.raw = 10 

#node.pdo.tx[4].trans_type = 254
TRANSMISSION_TYPE = node.sdo[tx4][2]
TRANSMISSION_TYPE.raw = 254 
```
But I dont know if add_variable() is useable without read() and save()

Furthermore, I'm sorry if my problem is obvious :)

christiansandberg, 2018-03-18:
I've made a fix if you want to try it. It is also possible to use `save()` without `read()` first. You just have to make sure to specify everything in your code.

morganPigeron, 2018-03-19:
I've tried your fix. It works partially. I wrote this : 
``` python
id = int(0x1400)
COB_ID = node.sdo[id][1]
TRANSMISSION_TYPE = node.sdo[id][2]

print COB_ID.raw
print TRANSMISSION_TYPE.raw
```
The print line for COB_ID worked unlike the other one.
Actually the COB_ID data type is uint 8 and the TRANSMISSION_TYPE is uint 32.

For now I'm using the ".data and unpack_from()" solution it works fine. I'm actually learning how to setup correctly the PDO.

Thanks !

christiansandberg, 2018-07-16:
The transmission type should according to the standard be UNSIGNED8 so if the node sends it as UNSIGNED32 that would be faulty behaviour. We've done quite a few workarounds now for non-compliant nodes so if we could discuss including this as well.

christiansandberg, 2018-08-13:
This should be fixed in v0.7.0.

closed 2018-08-13 by christiansandberg