canopen-python/canopen #65: Question - Object Dictionary, Ingenia Motion Drive

Wez50, 2018-04-14:
Hello,
I am looking to work with a servo drive from Ingenia motion control with CANopen. I have been able to put the drive into 'OPERATIONAL' mode and the PDOs are present on the Bus when i move the motor. (I have been decoding them with a picoscope).... also from Can-utils i am able to do a Candump, so i know data is able to come through my beaglebone. 

Ingenia has an object dictionary available for download, and i am using this. i am quite sure my issue is  down to the fact i've just started to learn this / don't have full application understanding, so i will be very greatfull for a kick in the right direction !

I keep getting traceback results saying "was not found in object dictionary" 

============
============
Traceback (most recent call last):
  File "HelloCan.py", line 7, in <module>
    device_name = node.sdo['VendorName'].raw
  File "/usr/local/lib/python2.7/dist-packages/canopen/sdo.py", line 176, in __getitem__
    entry = self.od[index]
  File "/usr/local/lib/python2.7/dist-packages/canopen/objectdictionary/__init__.py", line 61, in __getitem__
    raise KeyError("%s was not found in Object Dictionary" % name)
KeyError: 'VendorName was not found in Object Dictionary'
Exception in thread can.notifier (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 763, in run
  File "/usr/local/lib/python2.7/dist-packages/can/notifier.py", line 41, in rx_thread
  File "/usr/local/lib/python2.7/dist-packages/can/interfaces/socketcan/socketcan_ctypes.py", line 98, in recv
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'debug'

=========
=========

The program i have been using is below..... but this has now become a mess with my trying different things. I must have communication estabilished as i am able to put the drive into the OPERATIONAL state 

import canopen
network = canopen.Network()
network.connect(channel='can0' , bustype='socketcan')
node = network.add_node(20, 'object_dictionary.eds')
network.add_node(7, 'object_dictionary.eds')
network.nmt.state = 'OPERATIONAL'
device_name = node.sdo['VendorName'].raw
VendorID = node.sdo[0x1018][1].raw
#node.sdo['Producer heartbeat time'].raw = 1000

node.pdo.read()
node.pdo.tx[3].clear()
node.pdo.tx[3].add_variable('Application Status', 'Status All')
node.pdo.tx[3].add_variable('Application Status', 'Position actual value')
node.pdo.tx[3].trans_type = 254
node.pdo.tx[3].event_timer = 10
node.pdo.tx[3].enabled = True

#node.pdo.rx[4].clear()
#node.pdo.rx[4].add_variable('Application Status', 'Command All')
#node.pdo.rx[4].add_variable('Application Status', 'Command Speed')
#node.pdo.rx[4].enable = True

network.nmt.state = 'PRE-OPERATIONAL'
node.pdo.save()
node.pdo.export('database.dbc')
network.nmt.state = 'OPERATIONAL'
node.pdo.rx[3]['Application Commands.Target position'].phys = 1000
node.pdo.rx[3].start(0.1)
#node.pdo.state = 'OPERATIONAL'
with open('output.txt', 'w') as f:
        for i in range(50):
                node.pdo.tx[3].wait_for_reception()
                speed = node.pdo['Application Status.Actual Speed'].phys
                f.write('%s\n' % speed)
#def print_speed(message):
#       print('%s recived' %(var.name, var.raw))
node.pdo.tx[3].add_callback(print_speed)
time.sleep(5)
node.pdo.rx[3].stop()

network.disconnect()

===================
===================

a sample of the .eds file i am using is below, and i assume when i use the example provided device_name_obj = node.object_dictionary['ManufacturerDeviceName']
vendor_id_obj = node.object_dictionary[0x1018][1]

the fact the ingenia .eds does not have 'ManufacturerDeviceName' in the file is my issue ?
I have been trying lots of different things.... like 'VendorName' as this is in the file. but still i am doing something wrong

===========
==========
 :- 
[FileInfo]
FileName=0x19_0x29C_0x127
FileVersion=1
FileRevision=28
EDSVersion=402
Description=EDS for emcl chip
CreationTime=10:06AM
CreationDate=04-23-2013
CreatedBy=Ingenia Motion Control
ModificationTime=12:34AM
ModificationDate=06-14-2017
ModifiedBy=Ingenia Motion Control

[DeviceInfo]
VendorName=Ingenia Motion Control
VendorNumber=668
ProductName=emcl
ProductNumber=25
RevisionNumber=295
OrderCode=25
Baudrate_10=1
Baudrate_20=1
Baudrate_50=1
Baudrate_125=1
Baudrate_250=1
Baudrate_500=1
Baudrate_800=0
Baudrate_1000=1
SimpleBootUpMaster=0
SimpleBootUpSlave=1
Granularity=8
DynamicChannelsSupported=0
GroupMessaging=0
NrOfRXPDO=4
NrOfTXPDO=4
LSS_Supported=1

[MandatoryObjects]
SupportedObjects=3
1=0x1000
2=0x1001
3=0x1018

[1000]
ParameterName=Device Type
ObjectType=0x7
DataType=0x0007
AccessType=ro
DefaultValue=0x20192
PDOMapping=0x0

[1001]
ParameterName=Error Register
ObjectType=0x7
DataType=0x0005
AccessType=ro
PDOMapping=0x0

[1018]
SubNumber=0x5
ParameterName=Identity Object
ObjectType=0x9

[1018sub0]
ParameterName=Number of Entries
ObjectType=0x7
DataType=0x0005
AccessType=const
DefaultValue=0x4
PDOMapping=0x0

[1018sub1]
ParameterName=VendorID
ObjectType=0x7
DataType=0x0007
AccessType=ro
DefaultValue=0x29C
PDOMapping=0x0

christiansandberg, 2018-04-15:
There does not seem to be an index named VendorName. Maybe you are trying to read the information in the DeviceInfo section?

Wez50, 2018-04-17:
Ok, ive worked out that as i can set all the TPDO and RPDO items in the dive externally within Ingenia motion lab all i need to do is 'node.pdo.read() then node.pdo.save(). this then populates the canmatrix databse.dbc.  not sure why i can't overwrite them and save them with the library.

I have still been trying with no success, to get my motor to move to a position. by modifying the example code

# Start RxPDO4 with an interval of 100 ms
node.pdo.rx[4]['Application Commands.Command Speed'].phys = 1000

I have used:- node.pdo.rx[3] ['Controlword.Target position'].phys =1000
and i get Key errors saying its not in the map, but makes the same recommendation as a valid entry??  what is 'application commands' referring too ?

Im i correct in thinking that its cant be as simple as the example to move a motor ? 
would i have to set a value in the controlword to initiate the move / put the drive into the correct mode.

Target position is 0x607A , sub index 0x00 COB-ID 420 is set up in RPDO 3 in my drive and also
the controlword is set up , 0x6040, and the options are :
CMD_ENA_QUICK STOP = 0x00
CMD_DISABLE STOP = 0x06
CMD_ENABLE SWITCH ON = 0x07
CMD_ENA_OP    = 0x0F
CMD_CLR_ERROR = 0x80

I will be very grateful if you can point me in the right direction.
Wess

christiansandberg, 2018-04-17:
Looking at the Object Dictionary it seems like you should write something like this:

```python
node.pdo.rx[3]['Controlword'].raw = 0x0F
node.pdo.rx[3]['Target position'].raw = 1000
```

Or if you don't need PDOs you can just use SDO.

```python
node.sdo['Controlword'].raw = 0x0F
node.sdo['Target position'].raw = 1000
```

Code 0x06040041 means that some object could not be mapped to the PDO. You can increase the logging output to see where it fails.

```python
import logging
logging.basicConfig(level=logging.DEBUG)
```

christiansandberg, 2018-04-17:
You can also check out the 402 device profile support:
http://canopen.readthedocs.io/en/latest/profiles.html#cia-402-canopen-device-profile-for-motion-controllers-and-drives

Wez50, 2018-04-18:
Thank you for pointing me in the direction of the 402 information... i know its working as all the information goes to the drive.... but its not moving. 
so i am taking that up with Ingenia, as it might have something to do with a bit in the control word. as they reference a rising edge in the "New set point bit" and and "change set immediately" and i assume the status word will also acknowledge this.

but one quick question when you reference 'some_node' for example "network.add_node(some_node)" 
i have just used network.add_node(node) could this be a problem ?  

Thank you for all the help and advice
Wess

christiansandberg, 2018-04-18:
The `add_node()` method can take either a node number and an EDS-file or a `canopen.Node` object.

Wez50, 2018-04-18:
Thanks for confirming.
while i have this line of communication open with you i would like to ask your opinion on the best way to deal with sending multiple target positions. 
i have to wait for the acknowledgement bit in the status word before sending another target value to be stored in a buffer on the drive, before sending another control word. is it best to use some function like 'wait for reception' of the status word with an if statement for looking at the bit change before moving forward in the program? i have crudely used a delay between each target position to wait for the status word to change to get the positions into the buffer.

christiansandberg, 2018-04-19:
That sounds like the way I would do it. You can also use callbacks and a state machine.