canopen-python/canopen #66: No SDO Repsonse Received

donei003, 2018-05-15:
Hello,

First off, thank you for creating this CANopen library! This has helped me greatly. I am, however, running into an issue with access data from the Moog SmartMotor that I am currently using.

The current program I am running is as follows:
`import canopen
from canopen.profiles.p402 import Node402
import logging
import time
import pprint

logging.basicConfig(level=logging.DEBUG)
node = canopen.Node402(8, './SM5_0V3R4.eds')
network = canopen.Network()
network.add_node(node)
network.connect(bustype='socketcan', channel='can0', can_filters=None)

node.setup_402_state_machine()
node.sdo[0x6040].raw = 0x06
node.sdo[0x6041].raw
`
The error message (with debug) is:
`DEBUG:can:can config: {'interface': 'socketcan_native', 'channel': 'can0'}
INFO:can.socketcan.ctypes:Loading socketcan ctypes backend
INFO:can.socketcan.ctypes:Loading libc with ctypes
INFO:can.socketcan.native:Loading socketcan native backend
INFO:can.socketcan.native:Created a socket
DEBUG:can.socketcan.native:Binding socket to channel=can0
DEBUG:can.socketcan.native:Bound socket.
INFO:canopen.network:Connected to 'native socketcan channel 'can0''
INITIALISING
INFO:canopen.nmt:Sending NMT command 0x80 to node 8
DEBUG:can.socketcan.native:We've been asked to write a message to the bus
DEBUG:can.socketcan.native.tx:sending: Timestamp:        0.000000        ID: 0000    S          DLC: 2    80 08
INFO:canopen.nmt:Changing NMT state to PRE-OPERATIONAL
DEBUG:canopen.sdo:Reading 0x1800:1 from node 8
DEBUG:can.socketcan.native:We've been asked to write a message to the bus
DEBUG:can.socketcan.native.tx:sending: Timestamp:        0.000000        ID: 0608    S          DLC: 8    40 00 18 01 00 00 00 00
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/canopen/sdo.py", line 100, in read_response
    block=True, timeout=self.RESPONSE_TIMEOUT)
  File "/usr/lib64/python3.6/queue.py", line 172, in get
    raise Empty
queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test_canopen.py", line 13, in <module>
    node.setup_402_state_machine()
  File "/usr/local/lib/python3.6/site-packages/canopen/profiles/p402.py", line 57, in setup_402_state_machine
    self.pdo.tx[1].read()
  File "/usr/local/lib/python3.6/site-packages/canopen/pdo.py", line 238, in read
    cob_id = self.com_record[1].raw
  File "/usr/local/lib/python3.6/site-packages/canopen/common.py", line 64, in raw
    value = self.od.decode_raw(self.data)
  File "/usr/local/lib/python3.6/site-packages/canopen/common.py", line 24, in data
    return self.get_data()
  File "/usr/local/lib/python3.6/site-packages/canopen/sdo.py", line 240, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "/usr/local/lib/python3.6/site-packages/canopen/sdo.py", line 150, in upload
    with ReadableStream(self, index, subindex) as fp:
  File "/usr/local/lib/python3.6/site-packages/canopen/sdo.py", line 369, in __init__
    response = sdo_client.request_response(request)
  File "/usr/local/lib/python3.6/site-packages/canopen/sdo.py", line 118, in request_response
    return self.read_response()
  File "/usr/local/lib/python3.6/site-packages/canopen/sdo.py", line 102, in read_response
    raise SdoCommunicationError("No SDO response received")
canopen.sdo.SdoCommunicationError: No SDO response received
`

The EDS file I am using can be found [here](http://www.animatics.com/index.php?option=com_content&view=article&id=104&Itemid=9&dir=JSROOT%2Ftop+level%2F2.+Firmware+And+Fieldbus+Downloads%2FFieldbus+Config/CANOpen)

I have tried this both with and without the 402 profile so this is not the issue. The hardware setup is a SmartMotor connected to a Microchip CAN Bus Analyzer. At one point it seemed as if I was getting responses but all of a sudden I keep getting this error message. Any advice would be MUCH appreciated, thank you!

christiansandberg, 2018-05-15:
You should try to analyze the CAN bus traffic to see if messages are transmitted and the node responds to the requests. If not you may need to check connections, node ID etc.

donei003, 2018-05-15:
So I was able to get it working by disconnecting the USB-to-CAN adapter, setting up the can channel and then changing the node value to '1' instead of '8'. I'm not exactly sure what was the issue here however, before I was sending CAN packets to the motor and was unable to receive a response. At this moment I am able to receive packets and read data from the motor. 

In the response, it does say ```can_id=581, can_dlc=8``` can you explain what exactly these mean?

christiansandberg, 2018-05-15:
CAN ID 0x581 means that it is a SDO response (0x580) from node 1. DLC 8 is the data length in bytes of the message which is as expected.

donei003, 2018-05-16:
Ok, thank you so much! I was able to locate the ini file used during the motor's setup and the CADDR or CAN Address was 1 so that explains why it wasn't communicating in the first post. I have another question, however. I am trying to re-implement a PDO on the motor but I cannot seem to figure out how with the library. The PDO is as follows: 

[PDOMAP1]
DESC="Map RPDO1"
CINDEX=0x1400
MINDEX=0x1600
NUMENTRIES=4
MAP1=0x60400010
MAP2=0x60710010
MAP3=0x60870010
MAP4=0x60600008
TRANSTYPE=254
EVENTMS=50
COBVAL=0x201

The CINDEX, MINDEX, TRANSTYPE & COBVAL are fairly clear to me, however I am unsure how to implement the NUMENTRIES and EVENTMS side of things. Any help would be greatly appreciated. Thanks!

af-silva, 2019-10-16:
@donei003 Did you solve your problem? Can I close this issue

af-silva, 2019-10-23:
Closing the issue, if needed I can open it again. Thanks

closed 2019-10-23 by af-silva

srishtigusain11, 2021-11-23:
Hello,
I am facing similar problem- NO SDO response received

Traceback (most recent call last):
  File "C:/Users/gusains/Downloads/Test/CANOpen.py", line 47, in <module>
    print("The device type is 0x%X" % device_type.raw)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\variable.py", line 74, in raw
    value = self.od.decode_raw(self.data)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\variable.py", line 36, in data
    return self.get_data()
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\base.py", line 108, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 117, in upload
    fp = self.open(index, subindex, buffering=0)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 197, in open
    raw_stream = ReadableStream(self, index, subindex)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 243, in __init__
    response = sdo_client.request_response(request)
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 85, in request_response
    return self.read_response()
  File "C:\Users\gusains\Anaconda3\lib\site-packages\canopen\sdo\client.py", line 69, in read_response
    raise SdoCommunicationError("No SDO response received")
canopen.sdo.exceptions.SdoCommunicationError: No SDO response received
Bus error: an error counter reached the 'heavy'/'warning' limit

Following is the code I'm using -
```
try:
    # Start with creating a network representing one CAN bus
    network = canopen.Network()
    # CAN bus connect
    network.connect(channel='PCAN_USBBUS1', bustype='pcan', bitrate=125000)
    network.check()

    # Add some nodes with corresponding Object Dictionaries
    nodeId = 1
    file_path = os.path.join(os.path.dirname(__file__), 'sample.eds')
    # Add nodes to the network
    node = network.add_node(nodeId, file_path)

    for obj in node.object_dictionary.values():
        print('0x%X: %s' % (obj.index, obj.name))
        if isinstance(obj, canopen.objectdictionary.Record):
            for subobj in obj.values():
                print(' %d: %s' % (subobj.subindex, subobj.name))
    """
        for node_id in network:
        print(network[node_id])
    """
    # Simple variables
    # Read
    device_type = node.sdo[0x1000]
    print("The device type is 0x%X" % device_type.raw)

    # Write

    identity_obj = node.sdo[0x1018]
    node.sdo[0x1018].raw = 4000
    network.sync.start(0.1)
    print("Updated identity_obj is 0x%X:" % identity_obj.raw)
```