canopen-python/canopen #69: Understanding network.scanner.search

mohse-n, 2018-06-09:
I'm using Raspberry Pi with MCP2515 connected to a network consisting of two [Delta servo drives](http://www.deltaww.com/Products/CategoryListT1.aspx?CID=060201&PID=23&hl=en-US&Name=ASDA-A2). I am trying to get this example to work:
```
network.scanner.search()
time.sleep(0.05)
for node_id in network.scanner.nodes:
    print("Found node %d!" % node_id)
```
As far as I can tell, network.scanner.search sends SDO message to all possible nodes on the bus (1-128). 
```
for node_id in range(1, limit + 1):
            self.network.send_message(0x600 + node_id, sdo_req)
```
But how does it actually find out there are nodes present (which method in the class)? i.e. which method is listening for a response message from the node?

christiansandberg, 2018-06-09:
The `NodeScanner.on_message_received` method collects messages coming from slave nodes and is called from `Network.notify` which receives all messages coming in from the bus.

christiansandberg, 2018-06-10:
Yes that’s right.

mohse-n, 2018-06-13:
Thanks. How can I access the messages received by `Network.notify`?

christiansandberg, 2018-06-13:
You can create a [can.Listener](http://python-can.readthedocs.io/en/2.1.0/listeners.html) and attach it to `Network.listeners`.

```python
class MyListener(can.Listener):
    def on_message_received(self, msg):
        print(repr(msg))

network.listeners.append(MyListener())
```

closed 2018-06-14 by mohse-n