canopen-python/canopen #70: Manufacturer device-name object dictionary

mohse-n, 2018-06-14:
I'm trying to read the device name of a Delta A2 servo drive. Here's the relevant part in the `[eds](http://www.deltaww.com/Products/PluginWebUserControl/downloadCenterCounter.aspx?DID=4794&DocPath=1&hl=en-US)` file:
```
[DeviceInfo]
Vendorname=Delta Electronics,INC.
VendorNumber=0x000001DD
ProductName=ASDA-A2 Drive
ProductNumber=0x00006000
RevisionNumber=0x02000001
OrderCode=ASDA-A2
BaudRate_10=0
BaudRate_20=0
BaudRate_50=0
BaudRate_125=1
BaudRate_250=1
BaudRate_500=1
BaudRate_800=0
BaudRate_1000=1
SimpleBootUpMaster=0
SimpleBootUpSlave=1
Granularity=8
DynamicChannelsSupported=0
GroupMessaging=0
NrOfRXPDO=4
NrOfTXPDO=4
LSS_Supported=0
CompactPDO=0x00
```
I have tried:
`device_name = node.sdo['ProductName'].raw`
`device_name = node.sdo['DeviceInfo'].raw`
`device_name = node.sdo['DeviceInfo']['ProductName'].raw`
But for every single one of them I get the error 'was not found in Object Dictionary'.
The weird thing is, it can find the vendor ID index and subindex. i.e. running 
`vendor_id = node.sdo[0x1018][1].raw `
would not give me this error.

christiansandberg, 2018-06-14:
Manufacturer device name should be at index 0x1008 but that does not seem to be supported by this device.

mohse-n, 2018-06-17:
`TypeError: upload() takes exactly 3 arguments (2 given)`

christiansandberg, 2018-06-17:
Sorry, maybe I should look at [my own docs](http://canopen.readthedocs.io/en/latest/sdo.html#canopen.sdo.SdoClient.upload) first...

```python
node.sdo.upload(0x1008, 0)
```

mohse-n, 2018-06-17:
`canopen.sdo.SdoAbortedError: Code 0x06020000, Object does not exist`
So it appears the object does not exit in the device's object dictionary.

closed 2018-06-17 by mohse-n