canopen-python/canopen #71: Continuously updating value of an OD entry using RxPDO

mohse-n, 2018-06-18:
My plan is to update the position of a servo motor based on for instance value of sensor reading.
For giving position commands to a (Delta A2) servo drive, the manual specifies that I should update the value of the following entry in [the object dictionary](http://www.deltaww.com/Products/PluginWebUserControl/downloadCenterCounter.aspx?DID=4794&DocPath=1&hl=en-US)  with a RxPDO:
```
[60C1sub1]
ParameterName=Parameter1 of ip function
ObjectType=0x7
DataType=0x4
LowLimit=
HighLimit=
AccessType=rww
DefaultValue=0
PDOMapping=1
ObjFlags=0x0
```
I'm not completely familiar with capabilities of CANOpen and am not sure about how I should proceed. But the following steps comes to my mind:
1- I should first read the PDO configuration and the add this entry as a RxPDO:
```
node.pdo.read()
node.pdo.rx[1].clear()
```
In the example in documentation, the entry is add as such:
`node.pdo.rx[4].add_variable('Application Commands', 'Command All')`
How is this done in my case? I expected something like
`node.pdo.rx[1].add_variable[0x60C1][1]`
2- In the example in the documentation, a **fixed** value is sent to the node every 100 ms.
```
node.pdo.rx[4]['Application Commands.Command Speed'].phys = 1000
node.pdo.rx[4].start(0.1)
```
However,  what I'm trying to write looks like this:
```
while True:
     ##do some calculations and obtain the result
     node.pdo.rx[1][0x60C1][1].phys = result
```
and I can't figure out where `start(0.1)` would fit in this.

SquuuierMike, 2018-06-18:
You can start the transmission right after the mapping.

christiansandberg, 2018-06-19:
You need to call `start()` to start the periodic transmission but you may update the value at any time.

Try something like this:

```python
node.pdo.rx[1].read()
node.pdo.rx[1].clear()
node.pdo.rx[1].add_variable('Interpolation data record', 'Parameter1 of ip function')
node.pdo.rx[1].save()
node.pdo.rx[1].start(0.1)
while True:
    ##do some calculations and obtain the result
    node.pdo.rx[1]['Interpolation data record.Parameter1 of ip function'].phys = result
    time.sleep(0.1)
```

You may not need PDOs though. Try just using SDO and see if it works for you:

```python
while True:
    ##do some calculations and obtain the result
    node.sdo['Interpolation data record']['Parameter1 of ip function'].phys = result
    time.sleep(0.1)
```

mohse-n, 2018-06-19:
Maybe I should open a separate issue for this, but from `candump` output, it seems that I'm not sending any PDO messages.
```
node.pdo.rx[1].read()
node.pdo.rx[1].clear()
node.pdo.rx[1].add_variable('Interpolation data record', 'Parameter1 of ip function')
node.pdo.rx[1].enabled = True
node.nmt.state = 'PRE-OPERATIONAL'
node.pdo.rx[1].save()

node.pdo.rx[1].start(0.1)
node.nmt.state = 'OPERATIONAL'

node.pdo.rx[1]['Interpolation data record.Parameter1 of ip function'].phys = 0
time.sleep(5)
node.pdo.rx[1].stop()
```
`candump` output:
```
quad@quad-desktop:~/Desktop$ candump can0
  can0  603   [8]  40 00 14 01 00 00 00 00
  can0  583   [8]  43 00 14 01 03 02 00 00
  can0  603   [8]  40 00 14 02 00 00 00 00
  can0  583   [8]  4F 00 14 02 FF 00 00 00
  can0  603   [8]  40 00 14 03 00 00 00 00
  can0  583   [8]  4B 00 14 03 00 00 00 00
  can0  603   [8]  40 00 14 05 00 00 00 00
  can0  583   [8]  4B 00 14 05 00 00 00 00
  can0  603   [8]  40 00 16 00 00 00 00 00
  can0  583   [8]  4F 00 16 00 01 00 00 00
  can0  603   [8]  40 00 16 01 00 00 00 00
  can0  583   [8]  43 00 16 01 20 01 C1 60
  can0  000   [2]  80 03
  can0  603   [8]  40 00 14 01 00 00 00 00
  can0  583   [8]  43 00 14 01 03 02 00 00
  can0  603   [8]  23 00 14 01 03 02 00 80
  can0  583   [8]  60 00 14 01 00 00 00 00
  can0  603   [8]  2F 00 14 02 FF 00 00 00
  can0  583   [8]  60 00 14 02 00 00 00 00
  can0  603   [8]  2B 00 14 03 00 00 00 00
  can0  583   [8]  60 00 14 03 00 00 00 00
  can0  603   [8]  2B 00 14 05 00 00 00 00
  can0  583   [8]  60 00 14 05 00 00 00 00
  can0  603   [8]  2F 00 16 00 00 00 00 00
  can0  583   [8]  60 00 16 00 00 00 00 00
  can0  603   [8]  23 00 16 01 20 01 C1 60
  can0  583   [8]  60 00 16 01 00 00 00 00
  can0  603   [8]  2F 00 16 00 01 00 00 00
  can0  583   [8]  60 00 16 00 00 00 00 00
  can0  603   [8]  23 00 14 01 03 02 00 00
  can0  583   [8]  60 00 14 01 00 00 00 00
  can0  000   [2]  01 03
  can0  183   [2]  37 06
```
The last output corresponds to `node.nmt.state = 'OPERATIONAL'`. The next line, 
`node.pdo.rx[1]['Interpolation data record.Parameter1 of ip function'].phys = 0`
does not send anything on the bus.

acolomb, 2018-06-22:
Just a heads up: I just discovered that sending varying data using PDOs currently seems broken since version v.0.6.0. See also my [comment](https://github.com/christiansandberg/canopen/issues/72#issuecomment-399434901) on #72.

christiansandberg, 2018-07-16:
Should hopefully work with the latest master. Please re-open if you still have troubles making it work.

closed 2018-07-16 by christiansandberg