canopen-python/canopen #72: RxPDO is not sent

mohse-n, 2018-06-19:
From candump output, it seems that I'm not sending any PDO messages.

node.pdo.rx[1].read()
node.pdo.rx[1].clear()
node.pdo.rx[1].add_variable('Interpolation data record', 'Parameter1 of ip function')
node.pdo.rx[1].enabled = True
node.nmt.state = 'PRE-OPERATIONAL'
node.pdo.rx[1].save()

node.pdo.rx[1].start(0.1)
node.nmt.state = 'OPERATIONAL'

node.pdo.rx[1]['Interpolation data record.Parameter1 of ip function'].phys = 0
time.sleep(5)
node.pdo.rx[1].stop()
network.disconnect()

candump output:

quad@quad-desktop:~/Desktop$ candump can0
  can0  603   [8]  40 00 14 01 00 00 00 00
  can0  583   [8]  43 00 14 01 03 02 00 00
  can0  603   [8]  40 00 14 02 00 00 00 00
  can0  583   [8]  4F 00 14 02 FF 00 00 00
  can0  603   [8]  40 00 14 03 00 00 00 00
  can0  583   [8]  4B 00 14 03 00 00 00 00
  can0  603   [8]  40 00 14 05 00 00 00 00
  can0  583   [8]  4B 00 14 05 00 00 00 00
  can0  603   [8]  40 00 16 00 00 00 00 00
  can0  583   [8]  4F 00 16 00 01 00 00 00
  can0  603   [8]  40 00 16 01 00 00 00 00
  can0  583   [8]  43 00 16 01 20 01 C1 60
  can0  000   [2]  80 03
  can0  603   [8]  40 00 14 01 00 00 00 00
  can0  583   [8]  43 00 14 01 03 02 00 00
  can0  603   [8]  23 00 14 01 03 02 00 80
  can0  583   [8]  60 00 14 01 00 00 00 00
  can0  603   [8]  2F 00 14 02 FF 00 00 00
  can0  583   [8]  60 00 14 02 00 00 00 00
  can0  603   [8]  2B 00 14 03 00 00 00 00
  can0  583   [8]  60 00 14 03 00 00 00 00
  can0  603   [8]  2B 00 14 05 00 00 00 00
  can0  583   [8]  60 00 14 05 00 00 00 00
  can0  603   [8]  2F 00 16 00 00 00 00 00
  can0  583   [8]  60 00 16 00 00 00 00 00
  can0  603   [8]  23 00 16 01 20 01 C1 60
  can0  583   [8]  60 00 16 01 00 00 00 00
  can0  603   [8]  2F 00 16 00 01 00 00 00
  can0  583   [8]  60 00 16 00 00 00 00 00
  can0  603   [8]  23 00 14 01 03 02 00 00
  can0  583   [8]  60 00 14 01 00 00 00 00
  can0  000   [2]  01 03
  can0  183   [2]  37 06

The last output corresponds to node.nmt.state = 'OPERATIONAL'. The next line,
node.pdo.rx[1]['Interpolation data record.Parameter1 of ip function'].phys = 0
does not send anything on the bus.

mohse-n, 2018-06-20:
```

DEBUG:canopen.sdo:Reading 0x1400:1 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of COB-ID used by PDO (0x1400:1) is 515
INFO:canopen.pdo:COB-ID is 0x203
INFO:canopen.pdo:PDO is enabled
INFO:canopen.pdo:RTR is allowed
DEBUG:canopen.sdo:Reading 0x1400:2 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of transmission type (0x1400:2) is 255
INFO:canopen.pdo:Transmission type is 255
DEBUG:canopen.sdo:Reading 0x1400:3 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of inhibit time (0x1400:3) is 0
INFO:canopen.pdo:Inhibit time is set to 0 ms
DEBUG:canopen.sdo:Reading 0x1400:5 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of event timer (0x1400:5) is 0
INFO:canopen.pdo:Event timer is set to 0 ms
DEBUG:canopen.sdo:Reading 0x1600:0 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of number of mapped application objects in PDO (0x1600:0) is 1
DEBUG:canopen.sdo:Reading 0x1600:1 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of PDO mapping 1 (0x1600:1) is 1623261472
INFO:canopen.pdo:Adding Interpolation data record.Parameter1 of ip function (0x60C1:1) to PDO map
INFO:canopen.pdo:Adding Interpolation data record.Parameter1 of ip function (0x60C1:1) to PDO map
INFO:canopen.nmt:Sending NMT command 0x80 to node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: ï¿½
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
[... 87 lines cut ...]
DEBUG:can.socketcan.ctypes:Connected bcm socket
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: 
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
INFO:can.socketcan.ctypes:Sending BCM TX_SETUP command
DEBUG:can.socketcan.ctypes:Error sending frame :-/
INFO:canopen.nmt:Sending NMT command 0x1 to node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: 
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
INFO:canopen.nmt:Changing NMT state to OPERATIONAL
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:canopen.common:Writing Parameter1 of ip function (0x60C1:1) = 0
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.pdo:Updating Interpolation data record.Parameter1 of ip function to 00000000 in message 0x203
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:can.socketcan.ctypes:Error sending frame to stop cyclic message:-/

```

mohse-n, 2018-06-20:
```

DEBUG:canopen.sdo:Reading 0x1400:1 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of COB-ID used by PDO (0x1400:1) is 515
INFO:canopen.pdo:COB-ID is 0x203
INFO:canopen.pdo:PDO is enabled
INFO:canopen.pdo:RTR is allowed
DEBUG:canopen.sdo:Reading 0x1400:2 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of transmission type (0x1400:2) is 255
INFO:canopen.pdo:Transmission type is 255
DEBUG:canopen.sdo:Reading 0x1400:3 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of inhibit time (0x1400:3) is 0
INFO:canopen.pdo:Inhibit time is set to 0 ms
DEBUG:canopen.sdo:Reading 0x1400:5 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of event timer (0x1400:5) is 0
INFO:canopen.pdo:Event timer is set to 0 ms
DEBUG:canopen.sdo:Reading 0x1600:0 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of number of mapped application objects in PDO (0x1600:0) is 1
DEBUG:canopen.sdo:Reading 0x1600:1 from node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: @
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.common:Value of PDO mapping 1 (0x1600:1) is 1623261472
INFO:canopen.pdo:Adding Interpolation data record.Parameter1 of ip function (0x60C1:1) to PDO map
INFO:canopen.pdo:Adding Interpolation data record.Parameter1 of ip function (0x60C1:1) to PDO map
INFO:canopen.nmt:Sending NMT command 0x80 to node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: �
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
[... 87 lines cut ...]
DEBUG:can.socketcan.ctypes:Connected bcm socket
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: 
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
INFO:can.socketcan.ctypes:Sending BCM TX_SETUP command
DEBUG:can.socketcan.ctypes:Error sending frame :-/
INFO:canopen.nmt:Sending NMT command 0x1 to node 3
DEBUG:can.socketcan.ctypes:Packing a can frame
DEBUG:can.socketcan.ctypes:Data: 
DEBUG:can.socketcan.ctypes:Type: <type 'bytearray'>
DEBUG:can.socketcan.ctypes:sizeof frame: 16
DEBUG:can.socketcan.ctypes:Frame transmitted with 16 bytes
INFO:canopen.nmt:Changing NMT state to OPERATIONAL
DEBUG:can.socketcan.ctypes:Receiving a message
DEBUG:canopen.common:Writing Parameter1 of ip function (0x60C1:1) = 0
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:canopen.pdo:Updating Interpolation data record.Parameter1 of ip function to 00000000 in message 0x203
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:can.socketcan.ctypes:Trying to read a msg
DEBUG:can.socketcan.ctypes:Error sending frame to stop cyclic message:-/

```

christiansandberg, 2018-06-20:
Hmm, I suspected that.

```
DEBUG:can.socketcan.ctypes:Error sending frame :-/
```

It means that there was something wrong when setting up SocketCAN to do the periodic transmission. Do you have Python 3 perhaps? It will give you better error messages.

mohse-n, 2018-06-20:
Here's [the Python 3 log](https://www.dropbox.com/s/wcd5ie34b8lng8l/python3_log.txt?dl=0).

mohse-n, 2018-06-20:
And here's the candump output of Python 3. Am I sending empty RPDOs?
```
  can0  603   [8]  40 00 14 01 00 00 00 00
  can0  583   [8]  43 00 14 01 03 02 00 00
  can0  603   [8]  40 00 14 02 00 00 00 00
  can0  583   [8]  4F 00 14 02 FF 00 00 00
  can0  603   [8]  40 00 14 03 00 00 00 00
  can0  583   [8]  4B 00 14 03 00 00 00 00
  can0  603   [8]  40 00 14 05 00 00 00 00
  can0  583   [8]  4B 00 14 05 00 00 00 00
  can0  603   [8]  40 00 16 00 00 00 00 00
  can0  583   [8]  4F 00 16 00 01 00 00 00
  can0  603   [8]  40 00 16 01 00 00 00 00
  can0  583   [8]  43 00 16 01 20 01 C1 60
  can0  000   [2]  80 03
  can0  603   [8]  40 00 14 01 00 00 00 00
  can0  583   [8]  43 00 14 01 03 02 00 00
  can0  603   [8]  23 00 14 01 03 02 00 80
  can0  583   [8]  60 00 14 01 00 00 00 00
  can0  603   [8]  2F 00 14 02 FF 00 00 00
  can0  583   [8]  60 00 14 02 00 00 00 00
  can0  603   [8]  2B 00 14 03 00 00 00 00
  can0  583   [8]  60 00 14 03 00 00 00 00
  can0  603   [8]  2B 00 14 05 00 00 00 00
  can0  583   [8]  60 00 14 05 00 00 00 00
  can0  603   [8]  2F 00 16 00 00 00 00 00
  can0  583   [8]  60 00 16 00 00 00 00 00
  can0  603   [8]  23 00 16 01 20 01 C1 60
  can0  583   [8]  60 00 16 01 00 00 00 00
  can0  603   [8]  2F 00 16 00 01 00 00 00
  can0  583   [8]  60 00 16 00 00 00 00 00
  can0  603   [8]  23 00 14 01 03 02 00 00
  can0  583   [8]  60 00 14 01 00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  000   [2]  01 03
  can0  183   [2]  37 06
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
  can0  203   [4]  00 00 00 00
```

christiansandberg, 2018-06-20:
I can reproduce this but I have a hard time understanding what the hell is going on... But I think I've discovered a way to magically make it work as a temporary workaround.

```python
node.pdo.rx[1].start(0.1)
# Update with some dummy CAN data first...
node.pdo.rx[1]._task.update([1,2,3,4])
# ...now updating values seem to work for some reason
node.pdo.rx[1]['Interpolation data record.Parameter1 of ip function'].phys = 1
```

mohse-n, 2018-06-21:
Sending dummy frames seems to fix the issue, but unfortunately it only works on Python 3. I'm really hoping to use your library with ROS, which doesn't support Python 3 as of yet.
I also found out that sending [0,0,0,0] as the dummy frame has no effect, i.e. the succeeding frames will all have 0 data fields regardless of message specified in the code.

christiansandberg, 2018-06-21:
Not quite sure why it doesn’t work for Python 2. I have a pull request open to python-can to make the implementation for Python 2 and 3 basically the same, so that should hopefully work soon.

The update method only send an update if the new data differs, so that’s why it can’t be zeroes.

acolomb, 2018-06-22:
I just discovered that the current master breaks my PDO sending completely.  The data always stays the same.

Commit d24e10da045a2cde6c61f8577547c23322392503 is definitely to blame for that. After reverting it, python-canopen works fine.

Testing on Python 3.6.5 if that makes a difference.

christiansandberg, 2018-06-22:
I’ll try changing the implementation to always call modify_data() like before. It will not likely cause much overhead anyway.

christiansandberg, 2018-07-16:
This should work with latest master and python-can 2.2.0. If not please re-open.

closed 2018-07-16 by christiansandberg