canopen-python/canopen #75: Subscribers with the same COB ID are overwritten internally.

acolomb, 2018-06-20:
Let me first describe the scenario where the problem occurs.  I'm working on a system architecture that is not strictly a master-slave model.  Some slaves need to communicate among themselves, with the master (python-canopen) node just observing the communicated PDO data and setting up the appropriate mappings beforehand.

So the master configures node A's TxPDO1 mapping and sets a callback function to receive the data.  Then it configures node B's RxPDO1 mapping, overriding its COB ID record entry so that node B receives the PDO created by node A.  See the following example:

```python3
import canopen

network = canopen.Network()
network.connect(channel='vcan0', bustype='socketcan')

nodeA = canopen.Node(1, 'canopen/test/sample.eds')
nodeB = canopen.Node(2, 'canopen/test/sample.eds')
network.add_node(nodeA)
network.add_node(nodeB)

pdo = nodeA.pdo.tx[1]
pdo.add_variable('UNSIGNED8 value')
pdo.add_callback(lambda message: print(message))
pdo.enabled = True
pdo.save()

pdo = nodeB.pdo.rx[1]
pdo.cob_id = nodeA.pdo.tx[1].cob_id
pdo.add_variable('UNSIGNED8 value')
pdo.enabled = True
pdo.save()
```

After the last `pdo.save()`, node A's callback will never get called, because `canopen.Network.subscribe()` sets `nodeB.pdo.rx[1].on_message()` as the subscriber, overwriting nodeA's callback.  See also the [function's comment](https://github.com/christiansandberg/canopen/blob/b26721f2f46ddb06c24ebff6d52140badbab73a8/canopen/network.py#L59).

I see two possible solutions:

1. Allow `canopen.Network` to have multiple subscribers per COB ID.  This is not really needed in most cases, though, because the application does not need to get two callbacks for essentially the same received PDO.

2. Provide a way to inhibit the subscription in the second `pdo.save()` call, and error out if the same COB ID is subscribed to multiple times.

I have working code for both approaches and will open a PR shortly for the second method, which I would prefer.  Other opinions?

christiansandberg, 2018-06-20:
Allowing multiple subscribers would be really nice for the future so I'd go with that.

acolomb, 2018-06-22:
As stated in the [PR for solution 1](https://github.com/christiansandberg/canopen/pull/79#issuecomment-399428443), accumulating the callbacks can cause its own problems.  For me, this raises the question if multiple callbacks are actually ever needed?  Otherwise I still favor the second approach.

christiansandberg, 2018-06-22:
One example where this would be useful is the NMT message 0x000 which is common for all nodes. There could be some applications that might want to monitor specific messages also.

PR #79 (2018-06-25): Accumulate COB subscribers into a list.
The subscribers list was just a dictionary mapping each COB-ID to a
single callback function.  Extend that to use lists of callbacks,
which are called in turn.

This allows for different PDO mapping objects to be saved with the
same COB ID, without overwriting each other's on_message() callback
subscription.

One can now configure cross-communication between nodes by re-using
e.g. one node's TxPDO COB ID for another's RxPDO and calling read() or
save() for both.

Fixes #75.
closed 2018-06-25 by christiansandberg