canopen-python/canopen #86: Curtis Instruments implementation incompatibilites

acolomb, 2018-06-26:
Once again, thanks @christiansandberg for merging my slew of PRs to cope with the partly broken CANopen implementation in Curtis Instruments' motor controllers.

I have two more branches needed to work with these controllers in my application:

* https://github.com/acolomb/canopen/tree/curtis/messed-up-mapping-entries
* https://github.com/acolomb/canopen/tree/curtis/fixed-length-pdo-mapping-array

Maybe it would be wise to keep them on a separate branch within the upstream repository.  I'm open for suggestions how to handle this, then I can make appropriate PRs.

christiansandberg, 2018-06-26:
Oh my god... 🤢 Well, a Curtis branch could be a possibility. I'll think about it.

christiansandberg, 2018-07-02:
Go ahead and start pull requests for these too. This will probably not be the last time various hacks will be needed to support specific devices, and even if it makes the implementation less clean it is better that it can be used in real life.

acolomb, 2018-07-05:
I'll try to come up with some more generic flags than `curtis_hack` and extend the comments then.  Can you think of a good place in the documentation to collect these non-standard quirks?

acolomb, 2018-07-05:
There you go. #88 could be merged as-is, while #89 might need some better naming and documentation.

acolomb, 2018-07-10:
Thanks @christiansandberg for merging both pull requests.  If I come up with more workarounds, I will open new separate issues.

closed 2018-07-10 by acolomb