canopen-python/canopen #92: Transmission Type

CatSailer1279, 2018-07-19:
Hi,
I'm currently trying to read a ZILA TSR-CAN-03 ([datasheet](http://www.zila.de/images/dokumente/canopenbus/TSL-TSR-CAN-03_Handbuch_DE.pdf) unfortunately only in German available) temperature sensor and having some problems with the data type transmission.

When reading a SDO, the sensor will reply everytime with 4 bytes, even when I'm reading a 8-bit value. 

```
SDO command byte: 0x40, Client command specifier: Initiate upload request
OD main-index: Communication profile area (0x1800)
OD sub-index: 0x00
Reserved: 00000000

SDO command byte: 0x42, Server command specifier: Initiate upload response, Expedited transfer
OD main-index: Communication profile area (0x1800)
OD sub-index: 0x00
Data: 05000000
```

Therefore I get a missmatch error between the eds and the received message. My current workaround is to change the datatypes in the eds.
Is there any way to avoid this workaround? I tried to read the transmission type from the SDO object 0x1400, but unfortunately this object is not available.

Thanks for your help!

christiansandberg, 2018-07-20:
Could you try the master branch?

christiansandberg, 2018-07-23:
I’m pretty sure it should be fixed. I’ll make a new release the coming weeks. If it doesn’t work, please re-open.

closed 2018-07-23 by christiansandberg