canopen-python/canopen #93: Using a custom backend

HpLightcorner, 2018-07-20:
Hey!

With some of the latest commits i got into troubles when using a custom backend as described in the documentation - I think the library now strongly depends on python-can’s CyclicSendTask. At least the way described in the documentation (Using a custom backend) does not work as the CyclicSendTasks are missing.

I would suggest to remove the custom backend (at least from documentation) and point out that it's quite easy to extend python-can (which has a software-based CyclicSendTask fallback) using entry-points.

Hopefully I did not duplicate any ongoing issues.

Regards!

christiansandberg, 2018-07-20:
Yes I agree. The cyclic functionality wasn’t there before so it was slightly easier then.

christiansandberg, 2018-07-20:
Same for the ability to use plugins in python-can so I should update the docs to describe how to write a plugin instead.

closed 2019-10-16 by af-silva