canopen-python/canopen #94: Problem with initial PDO configuration

omerw91, 2018-07-22:
Hello,

I tried to run the example in the canopen.readthedocs
And i encountered two issues:

1. when setting "node.pdo.tx[4].event_timer = 10" and try to run a "node.pdo.save" i get the following error:
  File "read_pdo.py", line 58, in <module>
    node.pdo.save()
  File "/usr/local/lib/python2.7/dist-packages/canopen/pdo.py", line 59, in save
    pdo_map.save()
  File "/usr/local/lib/python2.7/dist-packages/canopen/pdo.py", line 293, in save
    self.com_record[5].raw = self.event_timer
  File "/usr/local/lib/python2.7/dist-packages/canopen/sdo.py", line 201, in __getitem__
    return Variable(self.sdo_node, self.od[subindex])
  File "/usr/local/lib/python2.7/dist-packages/canopen/objectdictionary/__init__.py", line 117, in __getitem__
    raise KeyError("Subindex %s was not found" % subindex)
KeyError: 'Subindex 5 was not found'

2. using a monitor i can see SDO messages going but no PDO start running. I also dont see a SYNC message going out.

Any ideas?

Best regards,
Omer.

christiansandberg, 2018-07-22:
It doesn’t seem that your node supports event timer. Check the object dictionary.

Which version of python-can are you using? Other people seem to have similar issues with Python 2.7 and python-can < 2.2.0.

af-silva, 2019-10-16:
@omerw91 did you find an answer to your problem??

af-silva, 2019-10-23:
Closing the issue, if needed I can open it again. Thanks

closed 2019-10-23 by af-silva