canopen-python/canopen #95: add LICENSE.txt to MANIFEST.in

uwasmuth, 2018-07-23:
Hi,

I wanted to use your component within a yocto environment. Yocto requires the license file to be in the deployed package. 
Could you add LICENSE.txt to MANIFEST.in. That would make it much easier to use it with yocto.
I can also open a pull request if thats easier. 
Thanks in advance

Ulrich

christiansandberg, 2018-07-23:
No problem. I’m on vacation so if you happen to have some time to make a pull request then I can merge it immediately.

uwasmuth, 2018-07-24:
Done. (Its my first pull request, so hopefully i've done it right?)

Mit freundlichen Grüßen / Best regards

Ulrich Wasmuth

PHOENIX CONTACT E-Mobility GmbH
Development Electronics
Hainbergstraße 2
D-32816 Schieder-Schwalenberg
Tel.  +49 5235 3-43114
Fax  +49 5235 3-43891
mailto:uwasmuth@phoenixcontact.com
http://www.phoenixcontact-emobility.com

<author@noreply.github.com>
Datum:  23.07.2018 21:18
MANIFEST.in (#95)

Phoenix Contact E-Mobility GmbH

Sitz der Gesellschaft / registered office of the company: 32816 Schieder-Schwalenberg
USt-Id-Nr.: DE283825847
Amtsgericht Lemgo HRB 7920 / district court Lemgo HRB 7920
Geschäftsführer / Executive Board: Dipl.-Wirt.Ing. Oliver Karl Stöckl
Diese E-Mail enthält vertrauliche und/oder rechtlich geschützte Informationen. Wenn Sie nicht der richtige Adressat sind oder diese E-Mail irrtümlich erhalten haben, informieren Sie bitte sofort den Absender und vernichten Sie diese Mail. Das unerlaubte Kopieren, jegliche anderweitige Verwendung sowie die unbefugte Weitergabe dieser Mail ist nicht gestattet.
----------------------------------------------------------------------------------------------------
This e-mail may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

christiansandberg, 2018-07-24:
Thanks a lot!

closed 2018-07-24 by christiansandberg