canopen-python/canopen #98: Issue on Windows

pycan, 2018-07-27:
Hi,

Recently, I come cross CANopen. I started using it on linux and it is just worked fine. However, the same script does not show the same behavior on Windows. It always could not go further that changing the state from preoperational to operational. When I try receiving my pdo's, I got an error message that complains on the SDO that is not sent/received. Could you please tell me what do I need to go on with my design. Note that PCAN-USB is used (https://www.peak-system.com/PCAN-USB.199.0.html?&L=1).  I use socketcan drivers on linux, whereas peak can drivers on windows.

regards,
Hajm

christiansandberg, 2018-07-27:
Have you specified the correct bitrate in your connect() call?

pycan, 2018-07-27:
Hi Christian,

Thanks for your reply.

Now a bit further. The below error speared after collecting the first coming pdo's 

ERROR message:
===========
error(10038, 'An operation was attempt on something that is not a socket')

Many thanks in advance

pycan, 2018-07-27:
Thanks Christian,

It works now. The node idea was wrong

closed 2018-07-27 by christiansandberg