canopen-python/canopen #103: Error in first example scanning network

ivomans, 2018-08-10:
I'm trying to get familiar with your library, but unfortunately I got stuck while trying the first example in an error I don't understand.

My code:
```
import time
import canopen

network = canopen.Network()
network.connect(bustype='serial', channel='/dev/ttyUSB0', bitrate=250000)
network.scanner.search()
time.sleep(2)
for node_id in network.scanner.nodes:
    print("Found node %d!" % node_id)
```
Resulting in error:
```
$ sudo python try-canopen.py
Exception in thread can.notifier for bus "Serial interface: /dev/ttyUSB0":
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "build/bdist.linux-armv7l/egg/can/notifier.py", line 69, in _rx_thread
    msg = bus.recv(self.timeout)
  File "build/bdist.linux-armv7l/egg/can/bus.py", line 77, in recv
    msg, already_filtered = self._recv_internal(timeout=time_left)
  File "build/bdist.linux-armv7l/egg/can/interfaces/serial/serial_can.py", line 138, in _recv_internal
    arb_id = (struct.unpack('<I', s))[0]
error: unpack requires a string argument of length 4
```
I've got 1 device in the network which is transmitting packages every second, that I can watch from my PC.

christiansandberg, 2018-08-10:
It seems to be an issue with the serial communication. Unfortunately even if you make it work, the protocol doesn't support standard frames (11-bit), only extended (29-bit), so it won't work anyway. You will have to do some hacking in python-can. Preferably you could improve the serial interface in python-can by adding support for distinguishing between standard and extended frames.

ivomans, 2018-08-13:
So this is a problem of the serial driver only? For now I'll proceed with hardware with socketcan, that's working without a problem. Would be nice in future to have a fully working serial possibility as well.

christiansandberg, 2018-08-13:
Yes but I can't know on which end. There is probably very few using the serial interface in python-can so you probably need to fix it yourself. Or if possible you can change protocol to the more standardised  [slcan](https://python-can.readthedocs.io/en/stable/interfaces/slcan.html) interface ([API](http://www.can232.com/docs/can232_v3.pdf)).

closed 2019-10-16 by af-silva