canopen-python/canopen #114: Losing packets while listening to network

nikodem-czechowski, 2018-10-03:
Hi,

I am using canopen (0.7.1) to listen (mainly) to my devices. I have approx 10 of them in the network. They sent a hearbeat signal (1 byte) every second and a status (8 bytes) every 5 second. Moreover, they can send more signals, but this is not the case.

When monitoring my network using simple script:

` frame = canopen.read_can_frame()`
`if frame.can_id == (0x180 + id):`
`    print parse_pdo(frame.data)`

I am not getting al the packets, that are send my way. However, if I use candump or write a simple script using can library:

`frame = can.bus.recv()`
`if frame.can_id == (0x180 + id):`
`    print parse_pdo(frame.data)`

Then everything is okay.  I would stick with this (can library), but apart from listening to these kinds of messages I need to send them some messages from time to time, and then canopen is crucial.

christiansandberg, 2018-10-03:
What does read_can_frame do?

nikodem-czechowski, 2018-10-03:
Oh sorry, that was some kind of strange version. What is actually happening is:

`network = canopen.Network()`
`network.connect(bustype='socketcan', channel='can1', bitrate=100000)`

`frame = network.bus.recv()`

I hope now it is clear what is happening.

christiansandberg, 2018-10-03:
That would explain things. The library also reads messages from the CAN bus so if you do it as well there will be conflicts. You can use the can.Listener interface to get notifier on messages.

```python
def on_message(msg):
    if msg.arbitration_id == (0x180 + id):
        print parse_pdo(msg.data)

network.listeners.append(on_message)
```

nikodem-czechowski, 2018-10-04:
Thank you! Now it works like a charm 🥇

closed 2018-10-04 by christiansandberg