canopen-python/canopen #115: CAN node is being detected onlz once.

anantshukla42, 2018-10-13:
Hi,
I am using the code mentioned below to detect my CANnode. 
It detects the can node only once and after that when I run the command again node is not detected. 
I have to restart my computer to get the node detected.

#!/usr/bin/python3

import canopen
import time

network = canopen.Network()

node = network.add_node(0x32, '0x19_0x29C_0x127.eds')

network.connect(channel='PCAN_USBBUS1', bustype='pcan', bitrate=50000)
# print("Error code: ", node.sdo[0x603F].phys)
network.scanner.search()
time.sleep(1)

print("Found %d node(s)" % len(network.scanner.nodes))
for node_id in network.scanner.nodes:
    print("Found node %d!" % node_id)

christiansandberg, 2018-10-13:
Sounds like something related to PCAN or the node in question. Not sure I can help you much there. Does the rest of the code work like reading the error code?

anantshukla42, 2018-10-14:
Hi,

I am using the same hardware on windows side but not getting any error
message. But in Ubuntu side, it detects only one time
Best regards
Anant Kumar Shukla

anantshukla42, 2018-10-14:
I have to power-off the CANopen node and restart in order to get it
detected.
Best regards
Anant Kumar Shukla

christiansandberg, 2018-10-14:
You probably need to log the CAN bus to see what messages are actually being sent from the computer and which messages are sent from the node. That might give you a hint on where to start looking.

closed 2019-10-16 by af-silva