canopen-python/canopen #116: support to ethernet/canopen converter

pugliast, 2018-10-21:
Hi,
I am going to develop a python application to manage some electrical drivers using canopen.
I have not a usb/canopen converter but a ethernet/canopen converter. Is there any example to start with?

Thanks

christiansandberg, 2018-10-21:
Not quite sure what you mean. Is there some protocol over TCP? I also have a repo called python-can-remote which is a network bridge for CAN, maybe you can use that as an example on how to implement your own CAN transport over Ethernet?

pugliast, 2018-10-21:
Hi,
I am trying this:

import can
bus = can.Bus('34.212.22.233:1201',
              bustype='remote',
              bitrate=500000,
              receive_own_messages=True)
msg =can.Message (0x58A, (0x40, 0x69, 0x5f,0x0,0x0,0x0,0x0,0x0))
bus.send(msg)
msg2 = bus.recv(1)
print('ok: ',msg2 )
bus.shutdown()

Where 34.212.22.233:1201 is the ip and port address of the Canopen/Ethernet gateway.

Thanks

christiansandberg, 2018-10-22:
Sorry that doesn’t help much. What is in the other end? Is it some kind of product? Is it a computer with some CAN interface? If it is my python-can-remote server, what is the output of this script? I won’t be able to help you unless you provide more information.

pugliast, 2018-10-22:
Hi Christian, thanks for your support.

 

My task is to read some parameters from a Lenze driver 8200 using SDO telegrams. To access the driver I have to go through a Lenze EMF2180IB ethernet/canopen gateway. 

 

My python application should sends the SDO telegram to the EMF2180IB ip 192.168.1.xxx:22080 and waits for the answer.

 

Looking to https://python-can.readthedocs.io/en/stable/interfaces.html I wonder that there is any interface for such ethernet gateways. 

 

Using the module python-can-remote I was not able to communicate with EMF2180IB. 

If you need more information please do not hesitate to ask me. I am an electrical engineer and I have not so long experience with python. 

 

Thanks again

Francesco

christiansandberg, 2018-10-22:
There is no support for this particular device. You will have to implement the communication layer yourself. First thing is to find a specification of the proprietary protocol used.

pugliast, 2018-10-22:
Ok.  Thank you for the feedback and the info! Appreciated :)

closed 2018-10-22 by pugliast

iurly, 2019-01-18:
@pugliast did you ever manage to access the EMF2180IB ethernet/canopen gateway?
Is there some specification of the protocol available?
Thank you!