canopen-python/canopen #118: Block transfer packet drops

rcd09, 2018-10-24:
Hello Christian,
I am using canopen as the client asking for a block transfer of log data from a CAN-based application we are developing at Cummins.  Running my python program to extract the log through PyCharm on a Windows 7  PC.  

Here is the canopen code to extract the log:

`    try:
        node = init_can_bus(args)
    except canopen.SdoAbortedError as e:
        print_error_response(get_time(), args.index, args.subindex, "Write", e.code)
        exit()
    except canopen.SdoCommunicationError as e:
        print("{} {}".format(get_time(), e))
        exit()
    except Exception as e:
        print("{} {}".format(get_time(), e))
        exit()

    # open output file and targetted sdo as an input stream
    try:
        with open(file_path, 'wb') as log_file:

            print("Begin transfer...")
            in_stream = node.sdo[index][subindexes[log_type]].open('rb', block_transfer=True)

            # read until we got no more
            data = in_stream.read()
            while data:
                log_file.write(data)
                data = in_stream.read()
            in_stream.close()

    except Exception as e:
        print("{} {}".format(get_time(), e))

    node.network.disconnect()`

It is clear after running this many times, that the server is sending packets faster than the client can get them.  Even on small, several hundred byte transfers, the client will ask to retransmit.  I have attached the debug log output from my python application where you can see the canopen debug trace.  

Is there a way I can get the client to read faster?  Or is the issue on the server side where it is not retransmitting correctly?

[log.txt](https://github.com/christiansandberg/canopen/files/2512477/log.txt)

Thank you,
Richard Detweiler
Firmware Engineer
Cummins,Inc

christiansandberg, 2018-10-25:
You should probably try to increase the receive buffer. Add `rxFifoSize=256` or something to the `connect()` method.

christiansandberg, 2018-10-25:
You can also reduce the block size.

```python
in_stream = node...
in_stream.blksize = 16
```

rcd09, 2018-10-25:
I'll increase the buffer size as you suggest.  Block transfer is very untested on our server 3rd party provider emtas (https://www.emtas.de/) as well.  I'll let you know if I have any changes to suggest after wringing it out.

christiansandberg, 2018-11-21:
Were you able to find a solution to this?

closed 2018-11-21 by christiansandberg