canopen-python/canopen #119: Modification of PDO indexes before they are saved in self.map

gongliang2, 2018-10-26:
@christiansandberg 
Hi Christian and all contributors,

many thanks for your contribution, so that I can use your canopen to enable our can communication. It works well, after I changed some of your source code.

Basicly I adjust the PDO indexes a little:
[image]

Sorry, I am not a expert in Canopen. I don't understand really why 0x1A00 for RPDO and 0x1800 for TPDO are used as base index for index shifting when PDOs are saved in self.map. I used 0x1400 for RPDO and 0x1800 for TPDO as they are used when we create them. And I extend the search range a little, 0x200 instead 128. Here is also the question, should we actually use 0x400 for the range?

It will be very nice if you or some of you guys can take a look at it and give me some suggestions. I can also push my changes to a branch.

best wishes!

Liang Gong

christiansandberg, 2018-10-26:
It was @af-silva that added this additional way of accessing the PDOs. I guess it is a matter of taste if you want to use the index of the communication parameters or the maps. Maybe we can support both ways?

gongliang2, 2018-10-29:
with 
> both ways 

do you mean:
1. over self.object_dictionary and
2. over self.map ?

af-silva, 2018-10-29:
Ok, first things first:

@gongliang2 Look at the table below

[image]

The last column shows where the configuration parameters for the PDO's are, this is, where you can configure **Transmission Type**, **Inhibit time** and **Event time**,  which means CAN objects 0x1400 for the RPDO and 0x1800 for the TPDO.

The PDO's are used to map CAN objects that are broadcast to the network like UDP messages, they don't need to confirmed if they arrived at the destination or not. The CAN objects that are mapped in the PDO's are configured in the CAN objects 0x1600 for the RPDO and 0x1A00 for the TPDO.

So, we can obtain this table:

| PDO |  Configuration | Mapping |
|:----- |:-----:| -----:|
| RPDO 1 | 0x1400 | 0x1600 |
| RPDO 2 | 0x1401 | 0x1601 |
| RPDO n | 0x14xx | 0x16xx |
| TPDO 1 | 0x1800 | 0x1A00 |
| TPDO 2 | 0x1801 | 0x1A01 |
| TPDO n | 0x18xx | 0x1Axx |
___

Like @christiansandberg said, what I have done is that you can access the PDO's (tx and rx) in three forms:

`node.tpdo[1]` **or** `node.pdo.tx[1]` **or** `node.pdo[0x1A00] `
`node.rpdo[1]` **or** `node.pdo.rx[1]` **or**  `node.pdo[0x1600]`

They all will give you the same PDO mapping. Which means from there you can access any object that you have mapped into the PDO, lets say for instance the object **0x606C** (Velocity actual value) can be access like `node.pdo[0x1A00][0x606C]` or `node.tpdo[1][0x606C]`

You are right that the PDO's range could be 0x200, the max available, [link](https://www.canopensolutions.com/english/about_canopen/pdo.shtml). I set the maximum value to 0x80 but a can set it to 0x200, but ins most cases we use only four

> A CANopen node usually has four transmit-PDOs and four receive-PDOs, where one COB-ID is reserved for each of these PDOs. 

I will change the max range to 200.
I hope this helps.

commit 55ca02a9 (2018-10-31): Merge pull request #121 from Introsys/PDO_increase_mapping

Fixes #119
commit e7025aa7 (2018-10-31): Fixes #119
closed 2018-10-31 by christiansandberg